openrocket/openrocket #460: Why show internal stage numbers in rocket side view window?

JoePfeiffer, 2018-10-23:
To the best of my knowledge, the only place the internal stage numbers ("Stage 0", "Stage 1", etc) are exposed to the user is in selecting the active stages in the bottom window of the GUI.  Everywhere else -- the design tree view, the simulation plots, and the simulation data exports -- the stage names, either as assigned by the program or by the user, are shown.  Displaying stage names in this window would be more consistent, and would also finesse Issue #360, since the stage numbering would not be visible.

Technically a new feature (and we should be in a feature freeze!) but it's just a couple of minutes work.  PR coming shortly.

teyrana, 2018-10-23:
Do we have "external" stage numbers?

I'm only aware of the one numbering scheme, as implemented in
FlightConfiguration.

JoePfeiffer, 2018-10-25:
Of course, there are no external stage numbers.  What I'm wondering about is stage names as opposed to stage numbers.

When a user creates a rocket, the first stage created is named "Sustainer".  Any further stages are named "Stage";  users are free to rename all of these stages anything they want.  These stage names, and not the stage numbers, appear in the Rocket Design window.

When the user runs a simulation, the simulation branches are named after the stage names -- so the first branch is "Sustainer" and the second is "Stage" (unless the user has chosen to rename the stages, of course).

So far as I've found, the only place stage numbers, as opposed to names, are exposed to the user is in the StageSelector, selecting the stages to be visible in the RocketFigure in the lower window of the GUI. 
 It seems to me it would be more consistent, and also more intuitive, if we used stage names there as well as everywhere else.

It'll probably be clearer to just see what the change looks like; check out my JoePfeiffer/openrocket show-stage-names branch and try it to see the difference.

teyrana, 2018-10-25:
Oh, gotcha.

Yea, that makes sense.    I'll bet the original reason had something to do
with the dynamic button width, and layout- but that's fixable.

teyrana, 2018-11-04:
@JoePfeiffer nominate to close.

Your PR fixed this, right?