openrocket/openrocket #463: Side boosters don't rotate with rocket

JoePfeiffer, 2018-10-25:
In the atttached simplified rocket (changing the filename from .ork to .txt so github would let me upload it), if I view the rocket in either the side view or the rear view, rotating the rocket doesn't rotate the side boosters.  They do rotate appropriately in all three 3D views.
[side-stage.txt](https://github.com/openrocket/openrocket/files/2515852/side-stage.txt)

teyrana, 2018-10-27:
Please clarify: 
How are you rotating the rocket? is this with the slider to the right of the RocketFigure?

In unstable I see the boosters rotating -- but not the motors. (so, still a bug). 

Is this what you see?

JoePfeiffer, 2018-10-27:
I'm sorry, yes I meant the motors.   Apparently yesterday was not my day for typing what I meant!

And yes, I did mean with the slider on the left of the RocketFigure.

teyrana, 2018-10-27:
Great! Then its fixed :)

teyrana, 2018-11-04:
@wcravens @JoePfeiffer 
Nominate to close.  This should be fixed in alpha9.

commit 1768c6d8 (2018-11-03): [fixes #463] Motors now rotate in RocketFigure side-view, back-view
commit 2b78647f (2018-11-03): Merge pull request #466 from teyrana/fix_463_motor_display

[fixes #463] Motors now rotate in RocketFigure side-view, back-view