openrocket/openrocket #469: Alpha 9: 3D view is crazily magnified first time it is selected

neilweinstock, 2018-11-04:
Any click restores it to normal, and then it will stay normal while working with that file.  Example:
[image]

This may be relevant:
[image]

teyrana, 2018-11-04:
Well that's odd.  We also haven't changed much in the 3d code.  Can you reproduce this in 15.04?

neilweinstock, 2018-11-05:
Yes, it does happen in 15.03.  That is a new problem, almost certainly connected to the fact that I just recently upgraded to Mojave.

So this is a Mojave issue, somehow.

teyrana, 2018-11-05:
Yea, I've had technical issues with graphics on my Mac for a while, now.
And then the card just totally died... So I sympathize.

itshanney, 2020-04-06:
@neilweinstock is this still an issue, or can it be closed?

neilweinstock, 2020-04-07:
Now that you mention it, I don't see the problem in the latest builds (at least, not in the packaged build you sent me).  I guess we can close it!

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2020-04-07:
Awesome :) 

Glad to hear it's resolved :)

closed 2020-04-07 by teyrana