openrocket/openrocket #476: FreeformFinConfig: On initial open, auto-scroll should center to fin bounds.

teyrana, 2018-11-04:
Split from issue #472 

> The fin editor always opens up zoomed out to the parent component. Often, the means the the fin itself starts out very small, particularly if it's mounted to a long BT. For example:

neilweinstock, 2019-03-28:
Just tried this on a design I'm working on for a friend.  The "About" box says Alpha 12 on the version I'm using (also tried this on the PR525 test version).
[image]

Here, all the side fins have been combined into one giant fin to (hopefully) get the best drag estimation.  Upon opening, the finset editor is zoomed neither to the parent component *nor* the fin itself... just an arbitrary front portion of the fin.  Here's the file.

[freeform_zoom.zip](https://github.com/openrocket/openrocket/files/3018301/freeform_zoom.zip)

teyrana, 2019-04-01:
@neilweinstock 

Is that offset part of the bug? It looks different in the main window, than in the fin-point window :/

neilweinstock, 2019-04-03:
No, I think the offset is OK.  The main tube is actually two separate pieces (connected by a coupler in the real model) in the model the fins are connected to the rear tube.

So I believe that aspect is being shown correctly, the only problem is that the whole fin is not being shown.

commit 99f758c7 (2020-07-18): Merge pull request #686 from teyrana/476/fff-auto-scroll

[fixes #502][Fixes #476] Fixes auto-scrolling of FinPointFigure bounds.
closed 2020-07-18 by teyrana