openrocket/openrocket #480: system.ser should not be in git

JoePfeiffer, 2018-11-08:
The file swing/resources/datafiles/presets/system.ser is generated when OR is built.  As such, it shouldn't be in git.

Its presence as a tracked file causes annoyances when attempting to switch branches, though it's otherwise unimportant.

teyrana, 2018-11-08:
I agree the current solution is annoying, but isn't that where we store out
motor database? (And load it into the application?

If we remove it from git, well want to add that step to eclipse & intellij
builds... Or possibly just remove it from some ant builds?

Also:
There was talk of porting that motor database to something more amenable to
git.  That sounda like a bigger, more long term solution.

JoePfeiffer, 2018-11-08:
Ah, I didn't realize it wasn't already in the eclipse and intellij builds (I just build using ant from the command line).  Yes, it would have to be added to those.

All the same, it seems to me that we that in general we shouldn't have generated files in the repository.

teyrana, 2018-11-09:
>
> All the same, it seems to me that we that in general we shouldn't have
> generated files in the repository.
>

Oh, *for sure*!  I think we're in agreement here.

Furthermore, it sounds like we've progressed to discussing the best
solution-- what needs to be changed, and how many hours we have to spare.

Maybe the first step is that system.ser isn't regenerated unless explicitly
called by an ant invocation?  That would at least cut down on the annoyance.

>

JoePfeiffer, 2018-11-11:
Sure, ant can do it (since it only recompiles modified java files), it just takes someone who either knows or is willing to learn ant!  Hmmm... looking at a build log, it appears to depend on swing/build/classes/net/sf/openrocket/utils/SerializePresets.class, and all the files in swing/resources-src/datafiles/presets/.

Things may be deeper than I thought -- I just tried to reconstruct those files (since it looks like they're supposed to be built from the vendor files in .../rocksim-components) using the generate-orc-files target, and I got a class not found exception on net.sf.openrocket.preset.loader.RocksimComponentFileTranslator
I wonder how long it's been since *those* files were last regenerated?

OK, I want to leave this open, but it's sure not something we want to resolve before next release.

JoePfeiffer, 2019-02-10:
It turns out the motor database is in .../core/resources/datafiles/thrustcurves/thrustcurves.ser.  It is also generated (by running 'ant serialize-thrustcurves' in .../core) and so also should not be in git.  But, unlike system.ser, there's no obvious (obvious to me, anyway) way to determine when thrustcurve.org has had new thrustcurves added (or existing ones edited), so it's even stickier.

JoePfeiffer, 2022-02-12:
Ah.  system.ser is no longer used at all.  I'm not finding any sign of it after a switch to current unstable, so I think it's gone.

closed 2022-02-12 by JoePfeiffer