openrocket/openrocket #481: Fins on pods not factored into CP

neilweinstock, 2018-11-16:
On the latest Alpha (and previous ones as well), fins on pods do not seem to be factored into the CP calculation.  In the following design, the CP location does not change when I add pods with fins to the end of each fin:
[image]

Here's the ORK file.
[cp_fins_on_pods.zip](https://github.com/openrocket/openrocket/files/2590664/cp_fins_on_pods.zip)

neilweinstock, 2018-11-16:
Now that I just looked at this, the CG doesn't change either.  Will go back and look into that.

JoePfeiffer, 2018-12-14:
The basic problem with CP is that it's calculated for one of the pods, and then multiplied by the number of instances of that pod to calculate the effect on the rocket.  If there are fewer than three fins on the pod, the forces depend on the angle of the fins.  The CP reported in the window is the worst case; in essence, what's happening is it's finding an angle at which one of the pods has minimal influence and then using that to estimate the effect of all the pods. which means it ends up ignoring them completely in the CP display.

neilweinstock, 2018-12-14:
Ah, well that is certainly a problem then; what you've just described is not a useful implementation.  Pods with less than 3 fins are common.

Is it fixable?  Sounds like it will be a good amount of work.

JoePfeiffer, 2018-12-14:
Need to keep looking.  At first glance I think it'll be pretty straightforward (essentially, just iterate through the instances of a pod to calculate forces rather than work them for one pod and multiply).

JoePfeiffer, 2019-02-04:
Please check to see if my PR 517 addresses this.

neilweinstock, 2019-02-04:
Is that in Daniel's latest JAR that he just distributed?

JoePfeiffer, 2019-02-04:
No, it's a new PR against unstable this morning.

neilweinstock, 2019-02-04:
OK, in that case I need to wait for a jar file to test.

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2019-02-05:
The old jar file has a few issues you could test ;)

neilweinstock, 2019-02-05:
I shall, don’t worry. Sorry it’s taking me so long, will get on it tonight.

- Neil

—

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

neilweinstock, 2020-05-01:
This one has been fixed along the way, closing.

closed 2020-05-01 by neilweinstock