openrocket/openrocket #482: Fin shape editor lost its grid

neilweinstock, 2018-11-16:
In the midst of all the great improvements to the fin shape editor, I just noticed that it lost its grid.

15.03:
[image]

Alpha 9: 
[image]

Was this an intentional removal?  The grid was handy.

Note also that the root was shown with a thick grey line, vs. the 1-pixel rendering of the parent object in the new release.  Although the parent object is sometimes useful, the thick grey line contributed more towards readability.

teyrana, 2018-12-01:
Sooo, this fix worked at the time, and now that it's merged, it no longer does.  -.-
https://github.com/openrocket/openrocket/pull/487

teyrana, 2018-12-01:
Lightened alpha channel on grid-line colors, and resubmitted.

neilweinstock, 2018-12-11:
In Alpha 10 I'm seeing:
1) Black grid lines (should be much lighter; I expected them to be given the most recent comment)
2) 5" gridlines.  That is nigh-useless for small rockets.  Not sure if this is supposed to be auto-adapting to the current zoom level, or if it's locked at 5", or what.  In my testing here it seems to stay at 5" always.

Screenshot:
[image]

neilweinstock, 2019-02-05:
I see no change in alpha 11.  Grid is still 5", same darkness as root edge and parent object outline.

neilweinstock, 2019-02-07:
Grid looks better in the *real* alpha 11.

I would still like to see the parent object outline shown thicker, as it used to be in previous release version.  This makes it easier to visually separate from all the other action in the editor.  Not sure if that should be addressed here or in a separate issue.

neilweinstock, 2019-04-18:
Here's how it looks in latest alpha:
[image]

Grid is still fixed at 0.5", and outline of parent object is still very hard to distinguish from 5" grid lines.  Was that supposed to have changed?

neilweinstock, 2020-05-01:
My comments here are unchanged.  The old grid was much "smarter" than the new one, and the parent object outline needs to be more prominent, like in the old release.

hcraigmiller, 2020-12-04:
New issues created  #(835, #836, and #837) to resolve outstanding elements of this issue.

hcraigmiller, 2020-12-11:
I have broken down the remaining elements of this issue and created new issues for each individual element, #835 for the physical size of the grid, #836 for the grid color, #837 for the color of the fin "point" indicator boxes, and #838 for the color of the fin lines (to be visually distinctive from the grid lines). Unless @neilweinstock believes something has been missed, this issue should be closed.

neilweinstock, 2020-12-12:
New issues cover it.  Closing this one.

commit 4bd3e4ff (2018-11-26): Merge pull request #487 from teyrana/fix_482_fin_shape_grid

[fixes #482] Restores grid lines
commit 6f0957bb (2018-11-26): [fixes #482] Restores grid lines
commit 143afb0f (2019-01-26): Merge pull request #508 from teyrana/fix_482_fin_grid

Fixes #482 // Tweak // Adjusts FinPointFigure background grid to be more useful
commit 5a033851 (2019-04-10): Merge pull request #520 from teyrana/fix_482_grid

[Fixes #482][graphics] adjusts line color and width in FinPointFigure
closed 2020-12-12 by neilweinstock