openrocket/openrocket #484: Cannot launch openrocket on Linux Arch64

snrkl, 2018-11-24:
```
snrkl@rincewind:~$ uname -a
Linux rincewind 3.10.105-bsp-1.2 #2 SMP PREEMPT Sat Oct 27 19:24:05 IST 2018 aarch64 aarch64 aarch64 GNU/Linux

snrkl@rincewind:~$ which java
/usr/bin/java
snrkl@rincewind:~$ java --version
openjdk 10.0.2 2018-07-17
OpenJDK Runtime Environment (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.3)
OpenJDK 64-Bit Server VM (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.3, mixed mode)
snrkl@rincewind:~$ 

snrkl@rincewind:~$ cat /etc/lsb-release 
DISTRIB_ID=neon
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="KDE neon User Edition 5.14"

```

Console Output:
```
snrkl@rincewind:~$ java -jar ~/Downloads/OpenRocket-15.03.jar 
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$2 (file:/home/snrkl/Downloads/OpenRocket-15.03.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$2
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
Launched OpenRocket using java -j OpenRocket-15.03.jar

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.RuntimeException: Error starting OpenRocket
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40)
	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:32)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38)
	... 1 more
Caused by: java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1367)
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1342)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1480)
	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:76)
	... 6 more
Caused by: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.ConstructorInjectorStore.get(ConstructorInjectorStore.java:49)
	at com.google.inject.internal.ConstructorBindingImpl.initialize(ConstructorBindingImpl.java:125)
	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:521)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
	at com.google.inject.internal.FactoryProxy.notify(FactoryProxy.java:46)
	at com.google.inject.internal.ProcessedBindingData.runCreationListeners(ProcessedBindingData.java:50)
[... 263 lines cut ...]
  83      8.422 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:529)
  84      8.422 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
  85      8.423 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
  86      8.425 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
  87      8.426 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
  88      8.426 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
  89      8.427 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.SingleFieldInjector.<init>(SingleFieldInjector.java:41)
  90      8.428 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.getInjectors(MembersInjectorStore.java:125)
  91      8.429 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.createWithListeners(MembersInjectorStore.java:95)
  92      8.429 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.access$000(MembersInjectorStore.java:34)
  93      8.430 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:42)
  94      8.431 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:39)
  95      8.432 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.FailableCache$1.apply(FailableCache.java:39)
  96      8.433 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:549)
  97      8.434 STDERR (SwingExceptionHandler.java:49) 	... 45 more
  98      8.435 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.ClassNotFoundException: javax.script.ScriptEngineFactory
  99      8.436 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:466)
 100      8.437 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:566)
 101      8.438 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
 102      8.438 STDERR (SwingExceptionHandler.java:49) 	... 67 more
 103      8.439 INFO   (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
 104      8.445 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 105     13.168 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 106     13.207 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2018-12-04:
The current version of openrocket requires java8, and you appear to have java10 as your default.  If you also have java8 on your system, you should be able to run /path/to/java8/java -jar OpenRocket-15.03.jar

For example, on my Debian system, I can run
snowball:613$ /usr/lib/jvm/java-8-oracle/bin/java -jar /usr/lib/openrocket/OpenRocket-15.03.jar

kernelsmith, 2019-03-14:
yeah, this is kind of a dupe of #347. For some reason the max compatible version of 8 is not mentioned anywhere obvious so we all keep opening issues like this

gallopinggoose6, 2019-04-15:
Then we should close this issue and we make 347 a bigger priority @JoePfeiffer @snrkl. Also, in addition we should mention that Java 8 is the max somewhere, as kernelsmith was saying.

wolsen, 2020-03-21:
This is now fixed in unstable

teyrana, 2020-03-22:
@wolsen @snrkl 
Closed as resolved.

closed 2020-03-22 by teyrana