openrocket/openrocket #497: Alpha 10: spurious thick fin warning

neilweinstock, 2018-12-11:
In Alpha 10 I am seeing a spurious thick fin warning when fins are inside pods.  In this design for instance:
[image]

the fin extensions on the outside are attached to phantom tubes attached to pods.  They are 3/32" thick, definitely not thick.  Deleting those fins makes the thick fin warning go away.

[thick_fins.zip](https://github.com/openrocket/openrocket/files/2669606/thick_fins.zip)

teyrana, 2019-04-04:
So, it turns out, it's due to the Phantom body tube.  The warning is generated if the fin is thicker than half of it's parents radius.  This would be relatively easy to fix if we can decide what that measure should *actually* be.

Given that this case already involves non-physical components:  If some user is using phantom tubes, they're already well off the reservation.  Let's just assume they know what they're doing, and can fend for themselves.  This is what the proposed fix PR does -- disable warnings for fins on phantom tubes.
(see: https://github.com/openrocket/openrocket/pull/535)

see: https://github.com/openrocket/openrocket/blob/unstable/core/src/net/sf/openrocket/aerodynamics/barrowman/FinSetCalc.java#L104
``` java
	// Add warnings  (radius/2 == diameter/4)
	if (thickness > bodyRadius / 2) {
		warnings.add(Warning.THICK_FIN);
	}
```

neilweinstock, 2019-04-07:
It is a good change to ignore phantom body tubes when simulating.  I would apply it to zero-length as well; they are sometimes needed and often cause spurious diameter discontinuity errors (a separate issue from this one, but since phantom body tubes came up...

teyrana, 2019-04-10:
Ahhhh, to be exact, warnings =/= sim.

The simulation probably already trivially ignores zero length/zero diameter
components, but that may or may not still emit warnings.

Do you happen to have an .ork which throws the warning?

teyrana, 2020-05-02:
Appears to have evolved to #571

commit 0823de5e (2019-05-29): [fix #497] Disables Fin-Thickness-Warnings on Phantom Fins
commit 7acaaf18 (2019-05-29): Merge pull request #535 from teyrana/fix_497

[fix #497] Disables Fin-Thickness-Warnings on Phantom Fins
closed 2020-05-24 by teyrana