openrocket/openrocket #500: Alpha 10: fins on pods have no mass

neilweinstock, 2018-12-14:
As far as I can tell, all fins on pods have a mass of N/A.  First noticed in Alpha 10, undoubtedly existed earlier.

teyrana, 2018-12-16:
Could Not Reproduce; please provide an example or more specific instruction.

Tested with both Pods & boosters <=> Trapezoidal Fins, Freeformfins, EllipticalFins

neilweinstock, 2018-12-17:
Huh, interesting.  Design attached.  Here's what it looks like when I edit the Fins on Pods:
[image]

[cp_fins_on_pods.zip](https://github.com/openrocket/openrocket/files/2686320/cp_fins_on_pods.zip)

teyrana, 2018-12-17:
Are you referring to the c.g. Displayed in the lower left of the config window?  The one thats just displaying "N/A"?

For reference, the overall C.G. In the main window updates on changes to the fins, but if your reeferringto the config window, then I understand what you're saying.

neilweinstock, 2018-12-17:
Yes, I was talking about the N/A.

However, in my tests, those fins are factored into neither CG *nor* total rocket mass.  I set the material to Titanium and no change to anything is observed.

What are you doing to make the fins affect CG?

teyrana, 2018-12-17:
Oh, hm.. It might be the material, too. Let me check that.

(Also, the parts database is buggered on linux, so that's another possibility)

neilweinstock, 2018-12-17:
FWIW I’ve tried *many* of the materials, no difference.

- Neil

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2018-12-21:
Status update: the "phantom" tube seems to be the trigger.  Setting it to have non-zero dimensions causes the fin to calculate an actual mass.  Not sure why, yet.

neilweinstock, 2019-02-05:
Looks fixed in Alpha 11, closing.

commit 70b75376 (2019-01-02): [fixes #500] May now calculate CG for fins attached to zero-dimension mounts
commit c5733b9b (2019-01-02): Merge pull request #505 from teyrana/fix_500_cg

[fixes #500] May Calculate CG for fins on zero-dimension mounts
closed 2019-02-05 by neilweinstock