openrocket/openrocket #516: Alpha 10: H128 engine

thzero, 2019-01-28:
Ran the example TARC Payloader with an H128W engine and got the following.  Tried it on a couple of other rockets, same result.   Also same result with a G64.

Also tested on Windows 10, and even in my java 9 builds.... all with the same result.

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.gui.main.SwingExceptionHandler$InternalException: An exception occurred during the simulation
	at net.sf.openrocket.gui.main.SwingExceptionHandler.handleErrorCondition(SwingExceptionHandler.java:120)
	at net.sf.openrocket.gui.simulation.SimulationRunDialog$InteractiveSimulationWorker.simulationInterrupted(SimulationRunDialog.java:413)
	at net.sf.openrocket.gui.simulation.SimulationWorker.done(SimulationWorker.java:98)
	at javax.swing.SwingWorker$5.run(SwingWorker.java:737)
	at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:832)
	at sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112)
	at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:842)
	at javax.swing.Timer.fireActionPerformed(Timer.java:313)
	at javax.swing.Timer$DoPostEvent.run(Timer.java:245)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at org.GNOME.Accessibility.AtkWrapper$6.dispatchEvent(AtkWrapper.java:715)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:109)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at java.awt.Dialog.show(Dialog.java:1084)
	at java.awt.Component.show(Component.java:1671)
	at java.awt.Component.setVisible(Component.java:1623)
	at java.awt.Window.setVisible(Window.java:1014)
	at java.awt.Dialog.setVisible(Dialog.java:1005)
	at net.sf.openrocket.gui.main.SimulationPanel$3.actionPerformed(SimulationPanel.java:176)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
[... 322 lines cut ...]
4889     61.465 VBOSE  (BasicEventSimulationEngine.java:331) Handling event FlightEvent[type=ALTITUDE,time=64.71296014442223,source=Example TARC Payloader,data=[4.982932389159027;1.0852874728188737]]
4890     61.465 VBOSE  (BasicEventSimulationEngine.java:132) Taking simulation step at t=64.71296014442223 altitude 1.0852874728188737
4891     61.466 VBOSE  (BasicEventSimulationEngine.java:253) HandleEvents: current branch = Booster stage
4892     61.466 VBOSE  (BasicEventSimulationEngine.java:255) Obtained event from queue:  FlightEvent[type=GROUND_HIT,time=65.21296014442223,source=null,data=null]
4893     61.466 VBOSE  (BasicEventSimulationEngine.java:256) Remaining EventQueue = [FlightEvent[type=SIMULATION_END,time=65.21296014442223,source=null,data=null], FlightEvent[type=ALTITUDE,time=65.21296014442223,source=Example TARC Payloader,data=[1.0852874728188737;-2.8100765119363813]]]
4894     61.466 VBOSE  (BasicEventSimulationEngine.java:331) Handling event FlightEvent[type=GROUND_HIT,time=65.21296014442223,source=null,data=null]
4895     61.466 VBOSE  (BasicEventSimulationEngine.java:255) Obtained event from queue:  FlightEvent[type=SIMULATION_END,time=65.21296014442223,source=null,data=null]
4896     61.466 VBOSE  (BasicEventSimulationEngine.java:256) Remaining EventQueue = [FlightEvent[type=ALTITUDE,time=65.21296014442223,source=Example TARC Payloader,data=[1.0852874728188737;-2.8100765119363813]]]
4897     61.466 DEBUG  (SimulationRunDialog.java:453) END, setting progress
4898     61.466 DEBUG  (SimulationRunDialog.java:423) Setting progress to 100 (real 100)
4899     61.466 VBOSE  (BasicEventSimulationEngine.java:331) Handling event FlightEvent[type=SIMULATION_END,time=65.21296014442223,source=null,data=null]
4900     61.466 VBOSE  (BasicEventSimulationEngine.java:255) Obtained event from queue:  FlightEvent[type=ALTITUDE,time=65.21296014442223,source=Example TARC Payloader,data=[1.0852874728188737;-2.8100765119363813]]
4901     61.466 VBOSE  (BasicEventSimulationEngine.java:256) Remaining EventQueue = []
4902     61.466 VBOSE  (BasicEventSimulationEngine.java:331) Handling event FlightEvent[type=ALTITUDE,time=65.21296014442223,source=Example TARC Payloader,data=[1.0852874728188737;-2.8100765119363813]]
4903     61.466 INFO   (BasicEventSimulationEngine.java:93) <<Finished simulating branch: Booster stage    curTime:65.21296014442223    finTime:65.21296014442223
4904     61.466 DEBUG  (Simulation.java:370) Simulation: returning from simulator, simulation took 510ms
4905     61.581 DEBUG  (SimulationRunDialog.java:366) Method 3:  alt=430.2007183797264  apogee=450.20822935920336
4906     61.581 DEBUG  (SimulationRunDialog.java:423) Setting progress to 100 (real 100)
4907     61.581 DEBUG  (SimulationRunDialog.java:235) Progressbar value 50
4908     61.582 DEBUG  (SimulationRunDialog.java:377) Simulation done
4909     61.582 DEBUG  (SimulationRunDialog.java:423) Setting progress to 100 (real 100)
4910     61.582 DEBUG  (SimulationRunDialog.java:235) Progressbar value 50
4911     69.293 USER   (SwingExceptionHandler.java:220) User requested sending bug report
4912     69.295 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2019-02-05:
I'm having a hard time duplicating this, but I'm noticing something odd in the log file output -- the F50 seems to go straight from ignition to burnout without any time steps in between.  Could you do me a favor and tar up your /$HOME/.java/.userPrefs/OpenRocket directory and send it to me at joseph@pfeifferfamily.net?  Thanks.

thzero, 2019-02-06:
Tried it on a clean vm and same thing.

Here is zip of the OpenRocket directory under $Home/.java/.userPrefs...
[https://drive.google.com/open?id=18B0f7Np6JSmSIuCbyNMDMVL8CNbj98HP](url)

JoePfeiffer, 2019-02-23:
The thrustcurve import code wasn't robust enough, and could create motors in the database with two thrusts at a given time, which would break the thrust interpolation code.  Addressed by PR 519.

gallopinggoose6, 2019-04-15:
Should this problem now be closed then?
I'm also envisioning a system where instead of having preloaded engine information, you could download a set of information from your engine manufacturer, and then add it to OpenRocket. This way the users could stay up to date with the engine information without having to wait for a new release of the program, or is there an implementation of this already that I am not aware of?

teyrana, 2019-04-15:
Afaik, thats in the feature request list (and not yet implemented,
unfortunately.)

itshanney, 2020-04-08:
Closing issue as it was fixed by PR #519. Please re-open if issue occurs again.

closed 2020-04-08 by itshanney