openrocket/openrocket #524: Recovery device name isn't updated in Recovery tab

JoePfeiffer, 2019-03-09:
Working on a dual deploy design, I originally didn't override the default parachute names ("Parachute").  I went back and renamed them "Main" and "Drogue".  When I went to the Motors & Configuration -> Recovery tab, I saw the parachutes were still just named "Parachute".  When I saved the design and restarted OR, the names had been updated (is this just an example of Issue 329?).

teyrana, 2019-03-09:
>  (is this just an example of Issue 329?)

Probably.  no great in harm in having an specific, explicitly reported issue for it, though.

teyrana, 2020-05-17:
@JoePfeiffer 

Working through our backlog; I have tested this, and it seems to be updating as expected now. 

Unless you can you reproduce it, I'd like to close this issue.

JoePfeiffer, 2020-05-17:
Looks like it's still there -- just opened a design, changed the name of the drogue chute to "Test"
[image]
but when I look at the recovery tab it's still named "Drogue"
[image]

When I save the design, exit OR, get back in and reopen the design, the name has changed

teyrana, 2020-05-17:
Okay, thank you for claryfying.  I can reproduce it now.

So, it's not *always* happening-- it updates in the component-tree, but not in the configuration > recovery tab.  hmm.

JoePfeiffer, 2020-05-17:
That seems to be consistent -- it seems to always update in the component tree, and never in the configuration>recovery tab.

commit 94a97759 (2020-05-31): Merge pull request #674 from teyrana/524-recovery-device-update

[fixes #524][ui] Recovery Devices update their name in the 'Recovery' tab
closed 2020-05-31 by teyrana