openrocket/openrocket #527: fin tabs loaded incorrectly.

JoePfeiffer, 2019-03-11:
When saving and then reading a design with TTW fins, my fin tabs are only 1.969 inches long.  Looking at the .ork file I find <tablength>0.20319999999999994</tablength> (which is the 8 inches I set them to), so it looks like the problem is in loading the design.

JoePfeiffer, 2019-03-12:
When the design is saved, the tab properties are written to the file before the fin properties (in particular, the tab length is written before the fin root chord).  when the tab length is read, it is validated, which sets the length of the tab to the root chord; since the root chord hasn't been read yet, the tab length is set to the default value of the root chord.

teyrana, 2020-02-17:
Closed as resolved.

commit ef080418 (2019-04-10): Merge pull request #530 from JoePfeiffer/fix-527

Fix #527
closed 2020-02-17 by teyrana