openrocket/openrocket #528: Can't Open OpenRocket

Dill18, 2019-03-12:
Hello. Brand new file downloads will not open. My java version is up to date too. How can I open the program?

JoePfeiffer, 2019-03-13:
Are you getting any sort of error message?  Also, what OS?

Dill18, 2019-03-14:
No, the download simply wouldn’t open after a complete download. I was later able to change the program that it opens with to the Java SE runtime thing, but this led to a black window popping up for what seemed about half a second. I assume this was my PCs command window opening for a split second then closing. I was able to get the program to open after downloading the OpenRocket file from the dropbox link which I believe holds an .exe file. There just seems to be an issue with the .jar file. My system runs window 10.

teyrana, 2019-03-14:
This might be an duplicate of: 
https://github.com/openrocket/openrocket/issues/347

If you have java 9, 10, or 11, Openrocket won't run.

nshire, 2019-03-21:
Openrocket currently only works with Java 8 or below. 

If you are Ubuntu, make sure you have JDK 8 installed by running `sudo apt install openjdk-8-jre` .
Then, run OpenRocket by first specifying the java 8 install path `/usr/lib/jvm/java-8-openjdk-amd64/jre/bin/java -jar [InstallDir]/OpenRocket.jar `

neilweinstock, 2019-03-28:
Sounds like he got it running using my packaged installer, which implies he's on Windows.

The packaged installer is the simplest solution in these cases.

itshanney, 2020-04-06:
Closing issue as reporter appears to be working and was encountering known issue of OR not running on Java 9+.

closed 2020-04-06 by itshanney