openrocket/openrocket #531: Uncaught exception printing design

JoePfeiffer, 2019-03-15:
When I try to "Print/Export PDF" for the attached design, I'm getting an uncaught exception "IllegalStateException:  Attempted to add motor data with an invalid fcid".  The pdf file is created, but is empty.
[airframe.ork.zip](https://github.com/openrocket/openrocket/files/2971886/airframe.ork.zip)

wolsen, 2020-03-14:
I can confirm this is a problem in unstable. It was introduced by commit 04c0914, which refactored some of the mass calculation. Unfortunately, there's a couple of additional problems in this code - the stageMass is using the wrong value to get the mass for the rocket which results in a NPE and the drawing dimensions for the rocket shape are off, causing the rocket to not render in the resultant PDF. Chasing the second issue down right now.

wolsen, 2020-03-15:
@JoePfeiffer I've built a jar of the branch and placed it at https://www.dropbox.com/s/dixowade3f3apn2/OpenRocket-wolsen-issue-531.jar?dl=0 if you want to give it a spin.

JoePfeiffer, 2020-03-15:
It addresses the initial reported problem, but there are a couple of issues remaining:
-- the thrust to wt ratio column has a value of (inf):1 (where inf is actually the infinity symbol, not the word inf)
-- pages 5 and 6 of the generated PDF are blank.
The attached PDF was generated from the rocket design I uploaded with the initial problem report, and demonstrates this.
[airframe.pdf](https://github.com/openrocket/openrocket/files/4335343/airframe.pdf)

wolsen, 2020-03-15:
Thanks @JoePfeiffer. I believe that pages 5 & 6 are for the nose cone. I saw this after I pushed and built this while playing a bit more with it. I hadn't seen the infinity for the thrust to weight ratio.

I'll look at both of those issues.

wolsen, 2020-03-15:
Interestingly - I don't have infinity (∞) for the thrust to weight ratio in my sample PDFs. But I also don't have a cluster of 5 motors either (nor does it appear that you do either, but it is listed as such).

JoePfeiffer, 2020-03-16:
Ah, I hadn't noticed the five-motor clusters!  I wonder if it's pulling the number of configurations (which is five) to get the number of motors in each configuration?

When you say you don't have the infinity thrust:weight nor five-motor clusters, is that using the ork I uploaded?

wolsen, 2020-03-16:
No, its using my own test file. But using yours I also get the infinite thrust:weight ratio. The number of configurations is due to a change in the InnerTube.getMotorCount() logic. It previously used to report the number of motors based on the cluster configuration, which I think is the correct behavior. The current code reports the number of motor configurations; we both just happen to have 5 in our documents.

JoePfeiffer, 2020-03-16:
It's interesting that it seems to calculate differently depending on whether you're running a simulation or printing a PDF.  Sounds like a candidate for refactoring.

wolsen, 2020-03-20:
It seems the TransitionShapes used in the PrintableNoseCone isn't taking the scaleFactor into account. I added some scaling back just for the PrintableNoseCone case, but think there's probably a way to do this where the scaling is done within the PrintableNoseCone code rather than the TransitionShapes (though I'm failing to see that method at the moment).

Here's [a build](https://www.dropbox.com/s/fa8ri3f7pb280tu/OpenRocket-wolsen-issue-531-3.jar?dl=0) of my latest push - I think it resolves most of the printing problems.

wolsen, 2020-03-22:
@JoePfeiffer not sure if you saw - but a new build is available to test this. Would love your feedback.

JoePfeiffer, 2020-03-23:
Looking good now.  I'm not seeing any of the issues I mentioned before.

teyrana, 2020-03-28:
@JoePfeiffer can I take that as review? ;)

teyrana, 2020-03-28:
@wolsen 
Hey, this PR is having trouble building under intellij -- but it was quickly solved by rebasing to the tip of 'unstable'.   And from _*there*_ it looked good :)

Would you mind doing that, and re-pushing to the source branch, above?

commit 6dae530a (2020-03-29): Fix IllegalStateException when export/print

Correct the logic for detecting if a motor has an invalid
FlightConfigurationId and throw an IllegalStateException only in the
case that it has an error.

Additionally, fix the lookup for the stageMass. The wrong value was
passed to the MassCalculator.getCGAnalysis(..) which resulted in
a NullPointerException.

Fixes #531
commit 7dbbb6c7 (2020-03-29): Merge pull request #591 from wolsen/fix-print

[Fixes #531] Fix printing/export as pdf
closed 2020-03-29 by teyrana