openrocket/openrocket #532: Running OR from a Java code / Python scripting / CLI

yuriachermann, 2019-03-17:
In the view of the advantages of working with lines of code in an batch way (scripted in Java), I would like to know if there has been anybody able to run any kind of simulations of Open Rocket from a **Java code**, even of a simple rocket.

So I wanna ask the devs in charge of the project if there is an standard **example of code** somewhere, used to **start the software, load a project and run the simulation**?

I really appreciate any help you can provide.

-Yuri Achermann

teyrana, 2019-03-17:
To directly answer your question, there is no example code for running from the command line because OpenRocket frankly doesn't have any of the features which would make this useful: no CLI arguments, no standard file-based definition for simulation parameters, and no standard output for simulation results.

yuriachermann, 2019-03-19:
Thank you for the attention, @teyrana.

I understood the CLI situation. But does that restricts also a Java code possibility? Because I would prefer using its own code (API) to develop my simulations. Do you know if there is an example of how I can do this? Any simple application to run a specific simulation so I can build on?

Do you think something like that could work???  ↓↓↓

    orkFile = new File("filename.ork");
    OpenRocketDocument doc = null;
    GeneralRocketLoader loader = new GeneralRocketLoader(orkFile);

    try {
        doc = loader.load();
    } catch (Exception ex) {
        response.getWriter().print(ex);
    }

    doc.getSimulation(config).getOptions().setWindSpeedAverage(windSp);
    doc.getSimulation(config).getOptions().setLaunchRodAngle(launchRodAngle);

    try {
        doc.getSimulation(config).simulate();
    } catch (SimulationException e) {
        e.printStackTrace();
    }

    doc.getSimulation(config).getSimulatedData().getBranch(0).get(FlightDataType.TYPE_LONGITUDE)

teyrana, 2019-03-19:
> Do you think something like that could work??? 

Oh sure.  Of course it could work. 

I'm just pointing out that you'd be writing most of the interface code yourself.  Openrocket IS open source after all . . . I'd love to see more people use it :)

Take a look at `openrocket/core/`. it contains all the functional stuff : defining a rocket, defining simulation conditions, running a simulation, etc. File load, save.  Etc.

The swing GUI is just a way to call that core API.  You could absolutely write your own CLI to call that same API.  And as projects go, it wouldn't be too hard.

(Again no same code, as such-- not unless you count the `openrocket/swing/` or `openrocket/android/` projects ;)

wcravens, 2019-03-19:
There is a good interface for writing plugins
<http://wiki.openrocket.info/Plugin_system> and an interface for running
simulations with python
<http://wiki.openrocket.info/Scripting_with_Python_and_JPype>.

I would love to know if the documentation is up to date and if these
interfaces are being tested against changes since 15.03.  Please have at it
and report back any issues that you find in the documentation or
application.  This is an outstanding but underutilized feature of open
rocket.

Many thanks,

Wes

yuriachermann, 2019-03-20:
I appreciate your response, @wcravens !

I've already used the orhelper.py module trying to script with Python and it's outdated (a lot of errors using 15.03). In order to run it, I had to use an older version of OpenRocket (1.1.9) and made some modifications in the code... Still, it doesn't seem to be working for me at all, cause it prints "Starting openrocket" one time and never stops processing something after that (infinite loop maybe).
The code I wrote in Python:
```
import orhelper

with orhelper.OpenRocketInstance('OpenRocket-1.1.9-mod.jar'):
    orh = orhelper.Helper()
    doc = orh.load_doc('simple.ork')
    sim = doc.getSimulation(0)
    orh.run_simulation(sim)
    data = orh.get_timeseries(sim, ['Time', 'Altitude', 'Vertical velocity'] )
    events = orh.get_events(sim)
```

OpenRocket-1.1.9-mod.jar (modified): 
[OpenRocket-1.1.9.zip](https://github.com/openrocket/openrocket/files/2986455/OpenRocket-1.1.9.zip)

Because of this Python situation, I think I prefer to use OR's own code (API) and develop the simulations in Java actually. But I'm struggling to understand the steps and methods used to start the software and complete a simulation...

I don't think I really understand the plugin system, does it let me script using Java and work with OpenRocket as an API?? Or does it only provides additional functionality to the software?

Thanks again!

bkuker, 2019-03-21:
It is probably out of date, but check out
https://github.com/bkuker/dispersion

The file
https://github.com/bkuker/dispersion/blob/master/src/com/billkuker/rocketry/dispersion/Dispersion.java
might
be what you are looking for.

-Bill

dwhiteley, 2019-03-22:
I also want to interact with OpenRocket via python scripting. I kept trying with version 15.03 with just attempting to load the example pin fin rocket file, which is named "test.ork" in my test code below:
````
from jpype import *
startJVM(getDefaultJVMPath(),'-Djava.class.path=OpenRocket-15.03.jar')
orf = JPackage('net.sf.openrocket.file')
or_loader = orf.GeneralRocketLoader(java.io.File('test.ork'))
doc = or_loader.load()
````

When I run the script, I get the following results:

````
19:54:14.659 [main] INFO  n.s.o.document.OpenRocketDocument - Clearing undo history of net.sf.openrocket.document.OpenRocketDocument
@2d6e8792
19:54:14.660 [main] INFO  n.s.o.f.o.importt.OpenRocketLoader - Loading .ork file
19:54:14.661 [main] DEBUG n.s.o.rocketcomponent.Rocket - Firing rocket change event ComponentChangeEvent[nonfunc]
19:54:14.662 [main] DEBUG n.s.o.rocketcomponent.Rocket - Firing rocket change event ComponentChangeEvent[motor]
19:54:14.662 [main] DEBUG n.s.o.rocketcomponent.Rocket - Firing rocket change event ComponentChangeEvent[tree]
19:54:14.662 [main] DEBUG n.s.o.rocketcomponent.Rocket - Firing rocket change event ComponentChangeEvent[nonfunc]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
jpype._jexception.ExceptionPyRaisable: net.sf.openrocket.file.RocketLoadException: Exception loading file: test.ork , Exception load
ing stream: null
````

From this, it seems like it's able to open the file and start reading it, but it errors out for some reason. Is there some preemptive step that I'm missing? The old orhelper.py code had some preliminary code dealing with a Startup class and Startup2 class, both which no longer exists. Does that code exist somewhere else, and do I need to invoke it before loading a file?

dwhiteley, 2019-03-24:
To add to my previous comment: Looking at the openrocket source, there seems to be a static class (net.sf.openrocket.database.Databases) that seems to have a lot of info used by the program. I wonder if the lack of this class caused my issues.

I noticed in the project that @bkuker pointed to, he needed to call fakeMethod from that class to initialize that class. I was unable to instantiate that class using Jpype. I attempted both retrieving the Databases class using JClass, and by getting the database package and calling database.Databases.fakeMethod(). Both these attempts failed.

Unless I receive any additional guidance, I'm coming to the conclusion that accessing the current version of OpenRocket via JPype is simply not possible. Unless addressed, I would recommend deleting the wiki pages referring to scripting to not lead anyone else astray.

dwhiteley, 2019-03-27:
OK. Trying something new. Trying this out with Jython, and I got somewhere different. Still got the same errors loading the document file. So I attempted to load the Databases class. The errors I got back were verbose enough for me to realize that I needed to do all of that dependency injection garbage. However, when I tried to do that, I got to another head scratcher:
````
>>> inj = Guice.createInjector(gui,plug)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>

1) No implementation for java.util.Set<net.sf.openrocket.formatting.RocketSubstitutor> was bound.
  while locating java.util.Set<net.sf.openrocket.formatting.RocketSubstitutor>
    for field at net.sf.openrocket.formatting.RocketDescriptorImpl.substitutors(RocketDescriptorImpl.java:9)
  at net.sf.openrocket.startup.GuiModule.configure(GuiModule.java:50)

1 error
        at com.google.inject.internal.Errors.throwCreationExceptionIfErrorsExist(Errors.java:435)
        at com.google.inject.internal.InternalInjectorCreator.initializeStatically(InternalInjectorCreator.java:154)
        at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:106)
        at com.google.inject.Guice.createInjector(Guice.java:95)
        at com.google.inject.Guice.createInjector(Guice.java:72)
        at com.google.inject.Guice.createInjector(Guice.java:62)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)

com.google.inject.CreationException: com.google.inject.CreationException: Guice creation errors:

1) No implementation for java.util.Set<net.sf.openrocket.formatting.RocketSubstitutor> was bound.
  while locating java.util.Set<net.sf.openrocket.formatting.RocketSubstitutor>
    for field at net.sf.openrocket.formatting.RocketDescriptorImpl.substitutors(RocketDescriptorImpl.java:9)
  at net.sf.openrocket.startup.GuiModule.configure(GuiModule.java:50)

1 error
````
I'm not sure if this is an issue with using Jython, or I'm forgetting something else when performing this operation. Is there another class which extends AbstractModule that I'm missing in the call?

yuriachermann, 2019-04-14:
Hey, guys!

I've opened the source code on eclipse and wrote a code to start OpenRocket version 1.1.9 (that version was chosen because I had the orhelper.py module to guide me). Even if I have the will, I'm a beginner in java (and also in programming) and I'm having trouble with the parameter "SimulationListener" to be used in the simulate() method. Here is my code:
```java
    public static void main(String[] args) {
	
		//Starts software
		Startup.checkDebugStatus();
		Startup.initializeLogging();
		Startup.initializeL10n();
		Prefs.loadDefaultUnits();
		Startup2.loadMotor();
		Databases.fakeMethod();
		System.out.println( "OpenRocket was started!" );
		//working at this point!!
		
		
		//TODO Loads rocket project		
		Rocket rocket = new Rocket();
		OpenRocketDocument doc = new OpenRocketDocument(rocket);
		
		GeneralRocketLoader loader = new GeneralRocketLoader();
		File orJavaFile = new File("simple.ork");
		
		doc = loader.load(orJavaFile);
		
		
		//TODO Start simulation
		Simulation sim = doc.getSimulation(0);
		final ArrayList<SimulationListener> listenersArray = new ArrayList<SimulationListener>(1);
		sim.getOptions().randomizeSeed();
		sim.simulate(listenersArray); //method is not applicable for this argument
		
		
		//TODO Get simulation data
		final ArrayList<FlightEvent> events = new ArrayList<FlightEvent>();
		FlightDataBranch branch = sim.getSimulatedData().getBranch(0);
		double apogee = branch.getMaximum( FlightDataType.TYPE_ALTITUDE );
		
		System.out.println( "The apogee was " + apogee );
    }
```
There are other problems already in the doc instantiation line when I run it by parts. Maybe if somebody remember the 1.1.9 version, could try to help me in this...

I'm working hard on this with no advances for a while, but with a help of someone experienced maybe it could happen!

Huge thanks to everyone!!

not7cd, 2019-11-16:
Hi, I collected some of the previous work here https://github.com/not7cd/orhelper
I also believe that these tests could be useful for this task.
https://github.com/openrocket/openrocket/blob/4a874b6ba95554042f8cfc05947b3bc43abccdab/core/test/net/sf/openrocket/file/openrocket/OpenRocketSaverTest.java#L213
https://github.com/openrocket/openrocket/blob/1470a846739e303a42933baed42d5165143aaa65/swing/test/net/sf/openrocket/IntegrationTest.java#L355

feltorn, 2020-01-14:
Hi @yuriachermann ,
i am currently working on a project and need to access OR in the same way.
Had you make progress with your Code from the 14 of April?
I would be highly interested in that!

-Felix Tornow

yuriachermann, 2020-07-19:
> Hi @yuriachermann ,
> i am currently working on a project and need to access OR in the same way.
> Had you make progress with your Code from the 14 of April?
> I would be highly interested in that!
> 
> -Felix Tornow

Hello, @feltorn !
What is the kind of project you are working on?
I had some progress, but I haven't had a success overall...
We can discuss the ideas if you want

-Yuri Achermann

bkuker, 2020-08-11:
I've integrated OR into several other applications over the years, and
created some one-off command line utils using OR.

This is not a complaint, but every OR release has broken them and added
something new to deal with...

Here is a screenshot of OR designs loaded into my motor simulator, and a
screenshot of a plot of landing & crash locations over thousands of random
perturbations of a single OR design...

If I remember correctly, difficulties ranged from trying to assemble only
parts of OR, because something like Guava is baked into OR pretty tightly,
to difficulties with things like the motor database being very awkward for
programmatic manipulation.

Here for example is the circa 2009 grossness I had to do for for an
editable motor
https://github.com/bkuker/motorsim/blob/master/gpl/com/billkuker/rocketry/motorsim/visual/openRocket/OneMotorDatabase.java
a
motor database that only ever returns one motor

Again, no fault of the OR team, as this kind of use case has never been a
goal of the project.

It is a sure thing that neither of these will compile or run with any
modern OR, but they could provide some ideas.

https://github.com/bkuker/dispersion
https://github.com/bkuker/motorsim

I am not sure about the interest level in making the changes needed to make
it easy to integrate OR code into other apps, or the effort required, but
the above projects might help you.

-Bill

[image: image.png]
[image: image.png]
[image: image.png]

jimmiedave, 2020-11-13:
In the service of the ongoing Wiki cleanup, I'm probably going to move the Github wiki page on this scripting into the dev section. Its doppelganger on the openrocket.info wiki will remain (as long as it's decent information) with the user guide for now.

If anyone has an issue with that, let me know, please.

SilentSys, 2021-01-14:
I have just updated the orhelper script to work with OR 15.03 and turned it into a pip package:
https://github.com/SilentSys/orhelper
https://pypi.org/project/orhelper/

Since I'm using this for another project, I plan on expanding on it and maintaining it long term. Although orhelper is fully working with OR 15.03 now, I still want to port all the two other old example scripts seen on the wiki to the new orhelper script.

Let me know if I should update the wiki.

jimmiedave, 2021-01-14:
Yes please!

Extensions (if I read you correctly) are one area of OR I know little about. That section is more robust than many of the others, but I don't know how I'd tell if it were current or correct.

I have a single ask: please don't use MS-Word to format your entry. I spent a few hours last night cleaning up a crazy page in there (not extensions).

DC

jimmiedave, 2021-01-14:
Revise and extend here - I think I may not move that page to the developer section - it's a judgement call, but it seems to me (after living in the wiki on and off for a few weeks) that it's more about using OpenRocket than building, and if there's fresh content coming in there, yes, will keep it.

SilentSys, 2021-01-15:
Okay I've updated the information on the github wiki https://github.com/openrocket/openrocket/wiki/Scripting-with-Python-and-JPype

I'll copy the info over to the other wiki when I have time.

Out of curiosity, which wiki is considered to be the main one?

SilentSys, 2021-01-15:
I've updated the other wiki as well: http://wiki.openrocket.info/Scripting_with_Python_and_JPype#Scripting_openrocket_with_Python_and_JPype

jimmiedave, 2021-01-16:
This is very nicely done - thanks so much! And now with Pyhon 3! (yeah, I know, no alternative really, but big props for pulling it up!)

Namespace question: 'net.sf.openrocket.*' is "sf" for SourceForge? And now that we've detached from SourceForge, does that legacy detail cause any trouble?

hcraigmiller, 2022-04-03:
See also, Issue #362 and Issue #308.

SilentSys, 2022-04-03:
Can this issue be closed now seeing as the wiki has been updated with a working solution?

https://github.com/openrocket/openrocket/wiki/Scripting-with-Python-and-JPype
http://wiki.openrocket.info/Scripting_with_Python_and_JPype#Scripting_openrocket_with_Python_and_JPype
https://github.com/SilentSys/orhelper

closed 2022-04-03 by SiboVG