openrocket/openrocket #536: java.lang.NullPointerException FreeformFinSetConfig.java

donotstayhere, 2019-04-05:
here is error messages

```
java.lang.NullPointerException
	at net.sf.openrocket.gui.scalefigure.FinPointFigure.updateSubjectDimensions(FinPointFigure.java:366)
	at net.sf.openrocket.gui.scalefigure.AbstractScaleFigure.updateFigure(AbstractScaleFigure.java:178)
	at net.sf.openrocket.gui.configdialog.FreeformFinSetConfig.updateFields(FreeformFinSetConfig.java:376)
	at net.sf.openrocket.gui.configdialog.ComponentConfigDialog.componentChanged(ComponentConfigDialog.java:132)
	at net.sf.openrocket.rocketcomponent.Rocket.notifyAllListeners(Rocket.java:510)
	at net.sf.openrocket.rocketcomponent.Rocket.fireComponentChangeEvent(Rocket.java:470)
	at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:1765)
	at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:1777)
	at net.sf.openrocket.rocketcomponent.FreeformFinSet.addPoint(FreeformFinSet.java:116)
	at net.sf.openrocket.gui.configdialog.FreeformFinSetConfig$FinPointScrollPane.mousePressed(FreeformFinSetConfig.java:407)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
```

teyrana, 2019-04-05:
@donotstayhere 
Hey, thanks for the bug report! 

Would it be possible to find out which version of OpenRocket you were running?  
Also, if possible, the rocket file, and what you were doing when it threw the issue?

donotstayhere, 2019-04-05:
@teyrana 
version: unstable branch
When I try to form of splited fin, some errors occur

thank you

teyrana, 2019-04-05:
"splited fin"

Sorry, I'm not familar with those... which component did you add? Trapezoid fins or Freeform?

donotstayhere, 2019-04-05:
I split free form fin and edit fin shape then some error occurred.

wolsen, 2020-03-06:
I'm able to recreate this using the latest unstable branch by creating a new rocket with a nose cone and body tube, adding Freeform fins, splitting them, and then adding boosters. When increasing the booster count from 2 to 3, the NullPointerException in the stack trace provided is thrown (albeit the line is 368 not 366). It occurs because the parent for the finset is null. I'm not yet sure why the parent for the finset is null, but I can confirm this is how to create it.

teyrana, 2020-04-04:
Hmm. Reproduced.   Albeit, not in the same was other users.

1. Add Freeform Fins
2. Split Fin
3. open one of the newly-created fins
4. Adjust "Number of Fins: "
 
note: this applies to both the "old" adjusted instance, and the new split instance

teyrana, 2020-04-05:
Somehow the call to 'parent.getForeRadius()'  at 'FinPointFigure.java:368' is reference the orginal fin, before splitting, which is no longer valid.

commit d61d0e47 (2020-04-07): Merge pull request #617 from teyrana/fix/536-fin-npe

[Fixes #536] NullPointerException while editing FreeformFinSet Components
closed 2020-04-07 by teyrana