openrocket/openrocket #537: Ubuntu apt-get openrocket breaks

ilyas121, 2019-04-16:
I installed by just downloading the jar after I got this error, not sure why wget fails from the package manager.

`Total wall clock time: 4.2s`
`Downloaded: 1 files, 21M in 3.9s (5.40 MB/s)`
`cleaning up temporary directory /tmp/openrocket.fjvQnX8rT0 ...`
`ERROR: wget failed to download `
`Processing triggers for mime-support (3.60ubuntu1) ...`
`Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.2) ...`
`Processing triggers for man-db (2.8.3-2ubuntu0.1) ...`
`Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...`

morganrallen, 2019-08-17:
I was also getting this but retrying succeeded in downloading the package.

wolsen, 2019-12-06:
This looks more likely to be a problem in the Ubuntu infrastructure or the packaging itself and not specific to the openrocket package in Ubuntu. Instead, a bug should be filed at https://bugs.launchpad.net/ubuntu/+source/openrocket/+bug/1838454

wolsen, 2020-03-20:
I think this issue should be closed, as it isn't directly related to the OpenRocket itself @teyrana

teyrana, 2020-03-22:
I'm agree with @wolsen 
Besides, it sounds like it was resolved, anyway :)

closed 2020-03-22 by teyrana

amunizp, 2020-05-07:
No this was not resolved. http://wiki.openrocket.info/Downloading_%26_Installing says it can be installed via ubuntu apt command  and it does not. 

I have filed a bug: 
https://bugs.launchpad.net/ubuntu/+source/openrocket/+bug/1877385

 there is also an open question that is related:
https://answers.launchpad.net/ubuntu/+source/openrocket/+question/204576

JoePfeiffer, 2020-05-07:
Since the packaged version (if installing with apt or apt-get did work) wouldn't run with current Java if the install did work, and we have no guarantee someone will step forward to package our next release for Debian, are we better off removing these instructions from the wiki or marking them as "of historical interest only"?

teyrana, 2020-05-07:
If those are true, I would agree.  But iirc, it sounds like people just want to wait for the next release to update them

JoePfeiffer, 2020-05-07:
My recollection is different -- it seemed like some sort of snap/appimage/flatpak image was preferred.  But maybe a gentler "doesn't work right now, hopefully after next release" note?

teyrana, 2020-05-07:
That's a good middle ground.  We should bring it up again when a release is closer.

JoePfeiffer, 2020-05-07:
I've put my 15.03 AppImage up on the Releases page, and edited the Linux section of the wiki at http://wiki.openrocket.info/Downloading_%26_Installing#Ubuntu.2FDebian_Repository
to reflect the changes.