openrocket/openrocket #538: CP calculation broken in alpha12

neilweinstock, 2019-04-18:
Perhaps one of the phantom tube fixes went awry.  Same OR file, comparison with previous alpha that I've been using:
[image]

Here's the file.
[alcubierre_cranberry.zip](https://github.com/openrocket/openrocket/files/3095876/alcubierre_cranberry.zip)

teyrana, 2019-04-21:
:/ 
Yarp.  That's definitely a bug.  Will take a look at it.

teyrana, 2019-04-21:
The CP contributions from fins on the Pods look like they're over-contributing.

neilweinstock, 2019-04-22:
Interesting. Wouldn’t over-contributing have pulled the CP back rather than all the way forward?

- Neil

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

wolsen, 2020-04-26:
Latest code in unstable shows this not to be an issue any longer and I believe this is resolved, but @neilweinstock can you confirm (I used your Alcubierre you supplied on another bug)?

[image]

@neilweinstock, as a side note - this is a beautiful rocket design (and the build thread on rocketry forum was awesome). Would you be willing to include this as an example rocket? We have a very nice example by @teyrana already for pods, but I think this is a very good professional looking example that could also be included to show some of the features of the next release of OR.

teyrana, 2020-04-26:
@wolsen
This rocket is definitely sexier!  ...But it also uses an internal tube as a circular (orbital) fin, which is an unsupported use case.   Meaning, we don't prevent it, but we also don't recommend it, because it leads to inaccurate simulations.

If we wanted this to be an official example, we'd probably want to add the circular fin type first.

Not to mention, this throws simulation warnings as a matter of course.
And currently,  #497

neilweinstock, 2020-04-26:
I’d be happy to let it be used for that purpose, but definitely better not to include a design with an unsupported tail ring.  I could probably dig up another design with good pod use that doesn’t throw warnings.

Support for tail rings would be very very welcome.  I’m a wee bit disappointed there not in the release we’re working on right now, but definitely not going to propose to add anything right now.

Also: the fin decals are a bit wonky due to issue #386, which has not gotten any love since I filed it (hint).  Once that is fixed then the designs will need to be updated to work with the fixed OR.

  *   Neil

neilweinstock, 2020-05-01:
@wolsen  Sorry I never answered your question... yes, the CP bug seems to be fixed, I'll close this now.

closed 2020-05-01 by neilweinstock