openrocket/openrocket #541: Changing "Position Relative To" setting creates unexpected results

neilweinstock, 2019-04-25:
After discussion on the dev mailing list, I'm reporting this as an issue.

In the previous release version, if you changed the "Position Relative To" option, the offset value would be automatically adjusted so that the component stays in the same place on the rocket.

In the development versions of the new release, this has changed.  Now, when you change the "Position Relative To" option, the offset value is not adjusted, and so the component moves.  For example, fins previously positioned at the bottom of the rocket move to the top.

It is unclear if this was an intentional change or not.  Either way, it seems unanimous among those who gave their opinion that the old behavior was preferred.

teyrana, 2020-05-17:
@neilweinstock 

I think this got fixed somewhere along the line, already.  

Unless you can still reproduce this, let's close it :)

teyrana, 2020-05-24:
Closing as resolved.

closed 2020-05-24 by teyrana

neilweinstock, 2020-07-30:
@teyrana This one seems to have cropped up again.  I could swear I just checked it in the last few days, but using a current build it's definitely happening.

Is it possible to reopen this or should I file a new one?

teyrana, 2020-07-30:
New issue, please :)

Plus any screenshots needed to reproduce the issue.