openrocket/openrocket #542: Build Failed on Java 11

acxz, 2019-06-13:
I am trying to fix the openrocket package for ArchLinux. The package depends on java, but the java package for ArchLinux is on Java 11.
I went ahead and tried to build the `jar` file by running `ant jar`. This gave me a long list of warnings and errors. After digging around a bit, I noticed #464 . I also tried to `ant jar` thzero's branch but I came up with a long list of warnings and errors there as well. Is there a plan or estimated date on when openrocket will be updated to work with the latest Java version?

For a temporary fix, I am thinking about making Java 8 a version specific dependency for the ArchLinux package instead of Java, so that users can actually use it.

It would be great if openrocket could work with the latest Java version. Sadly it has been a long time since I have done any Java programming, but maybe I might be able to help.

acxz, 2019-06-13:
I will keep this issue open since, getting openrocket to build on the latest java version is probably a goal for this project.

eduncan911, 2019-06-14:
Have you tried with OpenIDK?

acxz, 2019-06-14:
Yes all my building and running has been with openjdk versions of java.

eduncan911, 2019-06-14:
FYI, there's a PR to upgrade it to Java 9 and in the comments they said it builds on Java 11 and OpenJDK 10.5.

Maybe you can help resolve the conflicts so it can be merged?

acxz, 2019-06-14:
I did that check that branch, and it worked with java 10 but not on java 11 for me. If I do get time, I'll help out, but I have a couple other projects I am working on right now.

thzero, 2019-08-20:
It was my PR.  I've been meaning to check on it with OpenJDK 12 and/or the AdoptOpenJdk (https://adoptopenjdk.net/releases.html?variant=openjdk12&jvmVariant=hotspot).

I kinda paused on it, because there are some issues with the unstable branch - and it keeps getting features added to it, and revamps, but never actually released.

acxz, 2019-09-21:
Unstable fails on Java 12, I checked @thzero's `update` branch and I was able to build successfully.

acxz, 2019-10-12:
Now that Java 13 is out, I decided to try again, but the unstable branch still fails. @thzero's branch does build successfully on Java 13, however.

acxz, 2020-03-13:
After recent PRs I came back and tried the build on the default git branch (`unstable`) and it worked! Thx for the hard work guys!
Even though a new release hasn't been made, since the default git branch works, I will go ahead and close this issue.

closed 2020-03-13 by acxz