openrocket/openrocket #547: Ellipsoid Nose Cone Length Set to 0

WillMerges, 2019-10-05:
When the parameter for the Ellipsoid nose cone is set to zero the app crashes and freezes.

wolsen, 2020-03-13:
Which parameter are you setting to 0 to elicit this behavior? Also which version of OpenRocket are you running?

wolsen, 2020-03-15:
Ah, I think I've recreated this. It happens on 15.03 setting the length to 0 but doesn't happen on unstable. Easy recreate steps:

1. Create a new rocket
2. Add nose cone
3. Set length to 0

Program becomes unresponsive.

I was able to recreate this with ogive nose cone shape as well - I suspect its an issue with 0 length nose cone in general.

teyrana, 2020-03-15:
@WillMerges 
if we send you a jar file would you be willing to try it out?

WillMerges, 2020-03-15:
@teyrana I would be willing to try it out and attempt to recreate the problem if you send a jar.

teyrana, 2020-03-20:
Temporary download link:
https://drive.google.com/file/d/1chjHpCiH_S2xAEGU4AZd5X4CrWJFsi23/view?usp=sharing

WillMerges, 2020-03-20:
I just ran it, no issues with setting nose cone length to 0. I tried all shapes, with ellipsoid and ogive nothing hangs anymore.

teyrana, 2020-03-20:
Awesome!  Thanks :)
Meanwhile we're working on getting the next release out, do you can actually *use* these features!

(Closing as resolved)

closed 2020-03-20 by teyrana