openrocket/openrocket #550: Why freeform fin not updating instantly ?

Teals53, 2019-10-14:
in releases version (15.02) when i change something in freeform fin shape my static margin and fin type changing instantly in display but latest unstable build is not do this things why ?

Teals53, 2019-10-14:
btw why this new stage button added one more time ? its already exist here
[image]

neilweinstock, 2019-10-18:
1. I think your fit scale problem is the same as issue #502.  Please confirm.
2. The lack of real-time feedback in the freeform fin editor is known, although I can't find which issue it's in.  If I can't track it down I'll create a new one for that.
3. I'm not sure what you're referring to with regard to "nail button".

In many cases you are asking "why" the behavior is the way it is in the unstable version.  The short answer is because... it is unstable.  There are bugs.  If you don't want to deal with these bugs then use the release version.

Also: the video demonstrating bugs is helpful.  The language is not.  Please keep it civil.

Teals53, 2019-10-18:
1. maybe its same issue from #502 idk

2. you cant compare from 15.02 ?

3. sorry i write wrong i mean rail button :)

u say its unstable version its possible to have bugs okey but whats new in latest unstable if compare from latest release (15.02) ? i dont see any new things 

"Also: the video demonstrating bugs is helpful. The language is not. Please keep it civil."

i just cant find good word for explain this sorry xD

neilweinstock, 2019-10-18:
1. After looking a bit more I am pretty sure they're the same.
2. Not sure what you mean.  It is a known problem in the unstable, I'm just not sure which issue # is tracking it.  If I can't find it I'll create a new one.
3.  Ah, sorry.  Yeah, the default button size is.... a bit large. :)  I'll file a new one for that.

The new features are rail buttons, boosters, and pods.  Pods are the big one.

Teals53, 2019-10-18:
for second one i mean compare source code. rail buttons, boosters, and pods not enought for use latest unstable because latest unstable have lot of bugs etc and new features have this bugs too xD so 15.02 not have rail buttons boosters and pods but its not have this new bugs too so its more selectable version for work.

neilweinstock, 2019-10-18:
Yes, that is why the unstable is still in development.  If you want something stable use 15.03.

Teals53, 2019-10-18:
in my opinion latest unstable version need more features and less bugs bcs latest stable version released before 5 years ago but apparently its the opposite :/

Teals53, 2019-10-18:
openrocket not improving itself about all of them years the opposite is getting more bad

Teals53, 2019-10-18:
this is not normal under normal circumstances but maybe if its openrocket things change xD and its sad

closed 2020-03-19 by teyrana

Teals53, 2020-03-19:
@teyrana all of them this bugs are fixed ? so its your closing reason if not why u closed that ?

JoePfeiffer, 2020-03-20:
(1) An issue should be a single bug.  Not a whole collection of them.
(2) Frankly, the language in the bug report was such that I didn't bother looking at the bugs themselves.  Neil pointed this out to you.
(3) Your followups were not helpful in the slightest.  We know there are bugs, and we're working at fixing them (and there have been a *huge* number of fixes over the last couple of weeks).  Carping on the fact that an unreleased version has bugs keeping it from being doesn't do anybody any good.

In short, leaving it open wasn't useful.  We've got enough concise, clear reports to track down we didn't need this one.

Teals53, 2020-03-20:
ok

teyrana, 2020-03-20:
p.s. In the interest of being constructive:
https://developer.mozilla.org/en-US/docs/Mozilla/QA/Bug_writing_guidelines