openrocket/openrocket #552: why new unstable version not have some features from old version ?

Teals53, 2019-10-14:
isn't that supposed to be the opposite ?

neilweinstock, 2019-10-18:
My guess is that you're referring to Component Analysis and Rocket Optimization.  As far as I understand, those simply haven't been updated yet to work with the latest codebase.  There is no intent to leave them out in the long run, just no one has gotten to work on them yet.  There are still a lot of breakages to fix in the core functionality.

Teals53, 2019-10-18:
:/

gallopinggoose6, 2019-10-19:
Is it possible that some of these pull requests could be merged? I hesitate to work on anything as something I write might conflict with something else.

Teals53, 2019-10-19:
hm

JoePfeiffer, 2020-03-27:
Neil's guess is correct -- we've gone back and forth on whether to even try to have them back for next release, or if we should mark them as regressions to get a release out and then focus on getting them back.

gallopinggoose6 -- PR or issue?  There are only a few pending PRs, which I'm expecting to see committed very soon, and what they all do seems pretty clear.  As for issues, assign an issue to yourself and go for it.  That's something we've generally been pretty lax about but so far we've been pretty lucky about people not stepping on each other's toes.

gallopinggoose6, 2020-03-28:
Ok. I doubt I'll find time now, as things for me have changed, but I'll see what I can do.

teyrana, 2020-05-02:
This issue appears to refer to the Component Analysis Dialogue.

For updates specific to that feature, you can follow issue: #509; further progress and updates will be posted there.

closed 2020-05-02 by teyrana