openrocket/openrocket #553: angle number not changing ?

Teals53, 2019-10-14:


neilweinstock, 2019-10-18:
I *think* that 0 is just intended to indicate that the top of the slider represents 0 degree rotation.  I could be wrong.

That said, I think it would be a nice improvement if that value indicated the current rotation set by the slider.

Teals53, 2019-10-18:
> I _think_ that 0 is just intended to indicate that the top of the slider represents 0 degree rotation. I could be wrong.
> 
> That said, I think it would be a nice improvement if that value indicated the current rotation set by the slider.

but its working in 15.02 so maybe its just a bug .D

neilweinstock, 2019-10-18:
Sorry, you are correct, this is a breakage.

Teals53, 2019-10-18:
^^

wolsen, 2020-03-04:
From a git bisect, this appears to have been broken by commit f59ebdd0. It looks like it was a big change, so I need to look at it a bit.

teyrana, 2020-03-04:
Hmm, I thought we fixed this? 
Is this still presemt on unstable?

wolsen, 2020-03-04:
@teyrana yep - still present on unstable. It seems there's something to do with the update to the DoubleModel. The UnitSelector doesn't receive the events for the changes to the model so fails to repaint and update its text.

The UnitSelector adds itself as a changeListener to the model, however when the DoubleModel.setValue is invoked, it runs the code of setMethod.invoke(...) on the rocket figure and the event never gets back to the UnitSelector.

To me, it feels wrong that the UnitSelector registered for events and was never notified. A simple change would be to always call the fireStateChanged method inside of DoubleModel.setValue. However, I don't know this code well enough yet. The simple change here does indeed fix the problem, but I am not clear what all the side effects of such a change might be.

teyrana, 2020-03-04:
Blah. I'm probably thinking of  something else.

So, yea, I tried to cut down on number of fireStateChanged calls  (to reduce the flurry of messages) but it ended up causing more trouble than it solved.

wolsen, 2020-03-13:
A test jar is available at https://www.dropbox.com/s/j784mwjf9p7zhzm/OpenRocket-wolsen-issue-553.jar?dl=0

PR #575 (2020-03-14): Notify all event listeners when DoubleModel.setValue called
Previous refactoring of the event listeners lost notifications
to classes which care about changes to the DoubleModel. One
such example is the UnitSelector. Ensure that all the
ChangeEventListeners and StateChangeEventListeners are called
when the setValue method is invoked.

Closes #553
commit a2168ed8 (2020-03-14): Notify all event listeners when DoubleModel.setValue called

Previous refactoring of the event listeners lost notifications
to classes which care about changes to the DoubleModel. One
such example is the UnitSelector. Ensure that all the
ChangeEventListeners and StateChangeEventListeners are called
when the setValue method is invoked.

Closes #553
closed 2020-03-14 by teyrana