openrocket/openrocket #554: Default rail button size is ginormous

neilweinstock, 2019-10-18:
Default outer diameter and total height are 1 meter, which is big enough for a Falcon 9 I think. :)

[image]

(also there is no parts library for buttons at this point).

derived from issue #550.

wolsen, 2020-03-01:
@neilweinstock I've got a branch in progress that fixes the default rail button sizes, cleans up the rail button dialog, and adds some preset options. I should break them out into separate commits. Can you give this a try and let me know if it looks better for you?

neilweinstock, 2020-03-02:
I'd be happy to try it out.  Any way you could provide me with a jar file?

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2020-03-02:
Hey, guys, I've posted a jar on the referenced commit.

@neilweinstock Let us know what you find. 
@wolsen Thanks for fixing that stuff.  It's one of the many bits of cleanup we're trying to get together :)

neilweinstock, 2020-03-02:
Certainly looks better than before! :)

However, I do have questions and comments.  Here's what my window looked like after adding rail buttons to a body tube; refer to numbered callouts below:
[image]

1) I note that the side view visualizer does not seem to take the buttons into account when calculating extents.  Probably a separate issue.

2) Does the default size match an existing standard button?  I'm thinking one that would fit a 1010 rail.

3) Default material should probably not be Cardboard. :)

4) It dawned on me while looking at this that the icon on the button for adding Rail Buttons does not actually look much like a rail button, nor does it match the style of the rest of the icons.  Should be replaced.  Who does the icon work?  Also probably a separate issue.

teyrana, 2020-03-03:
1.  Yea, leta out that in a separate issue.   It's certainly true,  but that's neither a problem exclusive to this PR, nor a showstopper.

2.  Yes.  It is suzed for 1010 rail, according to the commit message.  (+1 for great commit messages, btw :)

3. Concur.

4.  Fair point.  I would advocate that's also a separate issue.

wolsen, 2020-03-03:
@neilweinstock thanks for testing and feedback!

@teyrana sure I can take a look at 3. I think the right approach is to add two new materials, Nylon and Delrin to the bulk materials list.

According to Wikipedia (and confirming via Dupont's material data center), Delrin appears to have a density of 1.42 g/cm^3 and Nylon has a density of 1.15 g/cm^3.

The overall mass estimate will still be off with the new materials as it doesn't account for any type of mounting hardware, but that's what the mass objects are for.

PR #570 (2020-03-03): Add railbutton presets
Allow rail buttons to define presets. This change doesn't actually add
any default presets to the library, but enables the ability to add some
preset options for rail buttons.

Partially Closes #554
commit 232b3637 (2020-03-03): Fix large rail button default size.

The default values assigned to RailButton sizes were configured with
a 1.0 meter outer diameter and 0.8 meter internal diameter (for 1010
rail buttons) which resulted in overly large rail buttons when added
using the default values.

Presuming that 1010 rail buttons are the common and an acceptable
default, change the default button sizes to align with the 1010 size.

Additionally, this change fixes some of the layout issues present in
add railbutton dialog.

Closes #554
commit 3995a330 (2020-03-03): Add new material types for rail buttons

Add two new material types for rail buttons, delrin and nylon.
The material densities are pulled from wikipedia and cross-referenced
with Dupont's material database.

Additionally, set Delrin as the default material for RailButtons.

Closes #554
commit f8993e89 (2020-03-03): Add rail button presets options

Allow rail buttons to define presets. This change doesn't actually add
any default presets to the library, but enables the ability to add some
preset options for rail buttons.

Partially Closes #554
closed 2020-03-03 by teyrana