openrocket/openrocket #560: bug in rocket specs at bottom of screen

Ross04, 2019-11-24:
[image]
The apogee, max velocity, and max acceleration are not working properly (see screenshot).
system: fedora 31. openrocket built from source
.ork file for replication purposes in zip file

[bugfix.zip](https://github.com/openrocket/openrocket/files/3883074/bugfix.zip)

teyrana, 2020-02-17:
@Ross04 Thank you for including the rocket file-- that's awesome :) !

Can you confirm which version of OR this is?  15.03, I assume? 

This is likely a dup of issue: #525 , and fixed in: #533 .  But it's hard to tell without further information.

Ross04, 2020-02-17:
Unable to confirm fixed, (for some reason it's not compiling for me now) .

> openrocket built from source
There is your OR version. I'll probably close this issue now since I am using your appimage most of the time now.

closed 2020-02-17 by Ross04