openrocket/openrocket #562: Include instructions to build from terminal

Houkime, 2020-01-20:
https://github.com/openrocket/openrocket/issues/340 points to http://wiki.openrocket.info/Developer%27s_Guide as a source for building instructions, but it is needlessly convoluted and tied to IDE (which i don't use).

On the other side, build.xml is just an Apache Ant build file.  
Actual build_from_terminal instructions are extremely simple:

### Building from source

* Install Apache Ant (e.g. `sudo pacman -S ant` for Arch Linux)
* Clone and build:
 ```
git clone https://github.com/openrocket/openrocket
cd openrocket
ant
```
* Run the .jar with `java -jar swing/build/jar/OpenRocket.jar` or copy it wherever you need

______________________________________
I am not sure about my exact wording and whether to include git itself in the required packages, but you can pretty much just add it to your README.md.

itshanney, 2020-02-13:
Note: You will need to ensure that you have JDK 8 and not newer. #347 is the issue that will deliver support for newer versions of Java, but that's not complete yet. In the meantime, ensure Java 8 is your JDK being used.

Your instructions are in essence correct. This is the basic procedure, assuming Linux, as you mentioned:
1. Install JDK 8. OpenJDK 8 is probably the most popular here.
2. Install Apache Ant and add to your PATH.
3. Git clone the code as you have done.
4. cd into openrocket folder
5. I would advise to start in core folder to test your setup
6. ant build

Wrapping that all up into a set of commands that can be used, give this a try. Sorry I don't use Arch Linux, but you can "translate" to your distro.

`
sudo apt-get install openjdk-8-jdk ant git
mkdir -p ~/Code
cd ~/Code
git clone https://github.com/openrocket/openrocket
cd openrocket/core
ant build
`

teyrana, 2020-02-15:
@Houkime @itshanney 

Friendly reminder:  Both of you have a the ability to edit the wiki pages. (if not, let me know-- that'd be a bug)

And this is exactly the sort of thing that would be useful to have documented there, and for the same effort as posting it here :)

teyrana, 2020-02-15:
As a starting point, I've copied over the comments here to a page: 

https://github.com/openrocket/openrocket/wiki/Instructions-to-build-from-terminal

Feel free to correct it, as you see fit, and if I've screwed up the editing, or misrepresented anyone, I apologize in advance!

itshanney, 2020-04-09:
Closing issue as wiki document has been updated and exists as a guide.

https://github.com/openrocket/openrocket/wiki/Instructions-to-build-from-terminal

closed 2020-04-09 by itshanney