openrocket/openrocket #563: Cannot Launch OpenRocket 15.03 on new OSX 10.13.6 running JRE 12.0.2

sandrobilbeisi, 2020-01-25:
System:
Mac OSX High Sierra: 10.13.6 (17G65)
Freshly installed JRE 12.0.2

this is basically a repeat of issue #433 but shows problem of all JRE's > v8

Error:

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.RuntimeException: Error starting OpenRocket
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40)
	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:32)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:567)
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38)
	... 1 more
Caused by: java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1367)
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1342)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1480)
	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:76)
	... 6 more
Caused by: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.ConstructorInjectorStore.get(ConstructorInjectorStore.java:49)
	at com.google.inject.internal.ConstructorBindingImpl.initialize(ConstructorBindingImpl.java:125)
	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:521)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
	at com.google.inject.internal.FactoryProxy.notify(FactoryProxy.java:46)
	at com.google.inject.internal.ProcessedBindingData.runCreationListeners(ProcessedBindingData.java:50)
	at com.google.inject.internal.InternalInjectorCreator.initializeStatically(InternalInjectorCreator.java:133)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:106)
	at com.google.inject.Guice.createInjector(Guice.java:95)
	at com.google.inject.Guice.createInjector(Guice.java:72)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at net.sf.openrocket.startup.SwingStartup.runInEDT(SwingStartup.java:143)
	at net.sf.openrocket.startup.SwingStartup.access$000(SwingStartup.java:43)
	at net.sf.openrocket.startup.SwingStartup$1.run(SwingStartup.java:79)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:303)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:389)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
[... 250 lines cut ...]
  84      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:529)
  85      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
  86      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
  87      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
  88      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
  89      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
  90      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.SingleFieldInjector.<init>(SingleFieldInjector.java:41)
  91      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.getInjectors(MembersInjectorStore.java:125)
  92      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.createWithListeners(MembersInjectorStore.java:95)
  93      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.access$000(MembersInjectorStore.java:34)
  94      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:42)
  95      2.302 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:39)
  96      2.303 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.FailableCache$1.apply(FailableCache.java:39)
  97      2.303 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:549)
  98      2.303 STDERR (SwingExceptionHandler.java:49) 	... 44 more
  99      2.303 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.ClassNotFoundException: javax.script.ScriptEngineFactory
 100      2.303 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:436)
 101      2.303 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:588)
 102      2.303 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
 103      2.303 STDERR (SwingExceptionHandler.java:49) 	... 66 more
 104      2.303 INFO   (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
 105      2.304 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 106      3.821 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 107      3.825 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

itshanney, 2020-02-13:
Sandro, this is a duplicate of #347 and is a known issue. OpenRocket needs Java 8 and efforts to upgrade have not been merged yet. Please try running with any Java 8 JRE and you should be good to go.

neilweinstock, 2020-02-17:
...or use the packaged installer: https://www.rocketryforum.com/threads/openrocket-packaged-installers-for-windows-and-mac-to-solve-all-your-java-problems.143540/

impressiver, 2020-02-17:
Another option is to install Java 8 using [SDKMAN](https://sdkman.io/) and run `sdk use 8.0.242-amzn` (or whatever variant you choose to install) to activate java 8 in the current shell.

acxz, 2020-03-13:
@sandrobilbeisi don't think its a duplicate issue of that since this that issue targets Java 9

teyrana, 2020-03-13:
Probably fixed by #572 .

@acxz 
Yes, the inducing jdks are different, but the underlying cause is the same.

closed 2020-03-19 by teyrana