openrocket/openrocket #568: Inconsistent CP and flight results when using pods or not

neilweinstock, 2020-02-17:
This image shows two outwardly identical rockets.  The one on the left, is a simple 3FNC.  The one on the right is too, except the fins are mounted to phantom body tubes on pods rather than directly on the main airframe.  The CP location and flight sim results are quite different between them.  Ideally should be the same.

This running on an Alpha build from April 2019, because later builds have been generally unusable for me.

[image]

[3fnc.zip](https://github.com/openrocket/openrocket/files/4216072/3fnc.zip)

JoePfeiffer, 2020-02-17:
This is the sort of issue that will be difficult or impossible to resolve.  There's aerodynamic interference between the fins and the body, which is accounted for in the simulation for the model on the left.  On the right, we don't have anything resembling the ability to discover the fins are in fact touching the body from their actual location without having it reflected in the tree, so the interference is not accounted for.

neilweinstock, 2020-02-17:
I thought that could be the reason.  Before giving up on it, though, is it certain that that is what accounts for the differences?  Forgetting Cd, do the stated effects at the fin/body joint also explain the difference in CP?

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

JoePfeiffer, 2020-02-17:
I'd expect it to.  Hmmm, this could probably be verified by temporarily taking the interference term out of the rocket on the left and see if it got the same answer as that on the right...

I"ll note the fin-body interference occurs in Sampo's thesis as Equations 3.55 and 3.56.

teyrana, 2020-02-17:
Well, the exact analogue should be replace the 3-fin-set. With 3 separate fin components of one each... That way, iirc, the interference should be ignored, even in 15.03.

neilweinstock, 2020-02-18:
> Well, the exact analogue should be to replace the 3-fin-set with 3 separate fin components of one fin each... That way, iirc, the interference should be ignored, even in 15.03.

Whoa... Did not know that and had to try it.  I took the 3FNC file from this issue and did a "Split fins" on the finset.  I saw no change in CP or in estimated altitude.  True in 15.03 and also in my alpha build.

When are you saying I should see a difference?

teyrana, 2020-02-19:
Ah. It turns out the [intereference corrections don't kick in until 5 fins](https://github.com/openrocket/openrocket/blob/da5dcca83df512f85766a8d83c7cbe676a03927e/core/src/net/sf/openrocket/aerodynamics/barrowman/FinSetCalc.java#L125)

So splitting four fins and less should calculate identificaly.  At 5+, you should see some difference.

JoePfeiffer, 2020-02-19:
That's fin-fin interference.  Body-fin interference is just below that in the code, and doesn't depend on number of fins.

JoePfeiffer, 2020-02-19:
That's fin-fin interference.  Fin-body interference is just below it in the code, and doesn't appear to depend on number of fins.

teyrana, 2020-02-19:
Hahaha, I was looking at that code, just the other day.

So, anyways, I stand corrected.  The split fins SHOULD (and do?) generate different results...

teyrana, 2020-02-19:
So, rewinding to the original question: does a single 3FNC match between 15.03, and the latest beta?

neilweinstock, 2020-02-19:
So, rewinding to the original question: does a single 3FNC match between 15.03, and the latest beta?
Yes, 3FNC with no pods seems to produce identical results.

Splitting the fins on either the old or new version has no effect on results.

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2020-02-19:
> Splitting the fins on either the old or new version has no effect on results.

Well that's a bug, isn't it? Maybe that's the issue here?   Assuming the body-interference term, what Cp should we expect?

neilweinstock, 2020-02-19:
Well from the *user’s* standpoint it’s a feature, or perhaps better called a lack of a bug.  There is absolutely no reason for me to assume that splitting the fins would affect the simulation.  I certainly never assumed it before, and nothing in the program suggests that this is the case.

I can’t say whether what’s happening behind the scenes right now is right or wrong or whatever.

If it truly is necessary to change the simulation results when fins are split, then a warning dialog should be displayed whenever doing that.

Is it not possible to calculate the fin/body interference for each fin individually?  I know zip squat about the physics involved, but it seems to me like each fin is operating independently in this case.

  *   Neil

wcravens, 2020-02-19:
Hi Neil,

It would be well worth you taking a quick spin through Sampo's technical
docs for the modelling and perhaps even the original thesis
<http://openrocket.info/documentation.html#>.  They are very digestible
without needing to understand all of the equations.  The motivation for all
simulation decisions and assumptions are in there though.  Sadly, we are
currently lacking the same clarity for the pod's design, or at least to the
best of my knowledge.

Wes

JoePfeiffer, 2020-02-19:
It is possible to compute the body-fin interference for each fin individually, and indeed (if I'm remembering correctly), in the case of a finset with three fins it calculates it for one fin and multiplies by three; with three separate fins it computes it three times and gets the same answer.

The problem with the second of the two original models that started all this (ie a 3FNC rocket implemented with three "phantom" pods, each with a single fin) is the simulation doesn't have the level of geometric reasoning needed to figure out that the fins are actually on the body.

I think at some point we simply have to assume a marginally sane designer:  rockets designed to fool the simulation will succeed in doing so.

teyrana, 2020-02-20:
Regardless, we may need to add more warnings about this, somewhere.

Users certainly want accurate time-of-flight, and downrange predictions, and this stuff goes towards make that more accurate.

>  There is absolutely no reason for me to assume that splitting the fins would affect the simulation. 

Honestly, that ones seems subjective to me.   If you change the rocket, you change the simulation.  And this wasn't even an option, until we added pods... 

Practically speaking, from the developer's point-of-view, it's A LOT easier to do the more complex calculations if all the fins are in one spot.   But I can see how one might not think of them as a set, but as "individual" fins, instead.

teyrana, 2020-02-20:
> The problem with the second of the two original models that started all this (ie a 3FNC rocket implemented with three "phantom" pods, each with a single fin) is the simulation doesn't have the level of geometric reasoning needed to figure out that the fins are actually on the body.

This is definitely true.  And while it's on our wishlist, it won't be anytime soon :/

neilweinstock, 2020-02-20:
Here's another try.
[image]

For this test, I split the fins half-way up, and put just the tips on the pods while leaving the bottom halves directly attached to the main airframe so they will have the fin/body interference accounted for (note that I still have not heard a definitive statement that the fin/body interference successfully accounts for the differences in the previous example).

In this test, the pods version has the CP is substantially further forward, and the altitude is once again greater.  If I had to guess, I'd think the CP should be about the same between these two, and the pods version should have higher Cd (and lower altitude) due to more fin edges (sim doesn't know the pod fins blend into the main fins).

neilweinstock, 2020-02-20:
> > There is absolutely no reason for me to assume that splitting the fins would affect the simulation.
> 
> Honestly, that ones seems subjective to me. If you change the rocket, you change the simulation. And this wasn't even an option, until we added pods...

I'm confused.  Ability to split fin sets has been there forever.  And splitting a fin set is not changing the rocket.  *And*, splitting the fin set does not seem to affect the simulation.  Are we talking about the same thing?

teyrana, 2020-02-21:
> Ability to split fin sets has been there forever. 

Yup. True.  The ability to split fins has been there for a while.  

The ability to add "phantom" pods? That's newer.

> And splitting a fin set is not changing the rocket. 

In the sense that changing the model in OR doesn't change the physical rocket on your workbench?  Yea, that's true.  

In the sense that changing the form of the model does or doesn't change the simulation in code? 
It may not be intuitive, but that can _absolutely_ change the outcome.    The rocket is not the model.  And the model is less accurate--as are ALL models.

> Splitting the fin set does not seem to affect the simulation. Are we talking about the same thing?

We _are_ talking about the same thing.  I'm saying that *according to the code* that split action SHOULD affect the simulation.  Either way, something is inconsistent.

JoePfeiffer, 2020-02-22:
IIRC the equation (and code) for the multi-fin set is simply the closed-form solution of the summation of several identical fins distributed evenly around the rocket.  So (without looking at the code -- things should finally settle down for me Monday) I'm not at all surprised to see the same result.

teyrana, 2020-02-23:
@JoePfeiffer 

If you have time to check out the code, recomend you check out:  [FinSetCalca.java](https://github.com/openrocket/openrocket/blob/da5dcca83df512f85766a8d83c7cbe676a03927e/core/src/net/sf/openrocket/aerodynamics/barrowman/FinSetCalc.java#L125), particularly lines 125-154, and 156-161.

JoePfeiffer, 2020-02-25:
It looks reasonable to me (though I didn't remember OR is using "classical", ie tech report, Barrowman and not Barrowman from his dissertation.

I need to work through the fin equations to really convince myself though...

JoePfeiffer, 2020-02-27:
Went back into Sampo's thesis to look up the handling of multiple fins -- the equation for the normal force for a single fin at some angle Lambda is 3.51, and is the force times sin^2(Lambda).  With a bit of trigonometry I was able to verify that the sum of three individual fins evenly distributed around the body (ie the split fin case above) is 1.5 times the force for one fin (Barrowman simplified this by assuming one fin is at an angle of 0 -- in this case, it contributes 0 and the other two contribute .75 each.  The assumption isn't necessary; putting them at theta, theta+120, and theta+240 gives the same total).

Equation 3.53 computes the sum of three or more fins evenly distributed around the body (ie the 3-fin finset) as N/2 where N is the number of fins.  So the 3-fin finset case also gives a total of 1.5 times a single fin.

So, in getting the same result for the split fin and the 3-fin finset cases, the code is doing the right thing.

Are you thinking the body-fin interference should be different in the two cases (as that's calculated in lines 156-161)?

teyrana, 2020-02-28:
@JoePfeiffer 
Ahh, thank you for the explanation, that explains the fin-fin interference.

I was also concerned about the fin body interference, and that it seems to adjust the notnal force, based on the body diameter, and the fin aspect ratio...

So moving those fins to phantom body tubes (vs mounted on physical body tubes) *ought to*, change the calculations, right? (Hopefully for the better?)

neilweinstock, 2020-03-03:
Has anyone looked at the second example I posted above?

JoePfeiffer, 2020-03-03:
In the case of fins on phantom pods it's calculating the body-fin interference based on the 0-diameter pod, and so calculates it incorrectly.  If I set the pod diameter to be very slightly greater than 0 it'll actually warn that thick fins may not be modeled accurately (the thick fin test checks for a 0-diameter body and doesn't put the warning up if that's the case.  Do we want to remove the check, so we'll get the warning on 0-diameter pods?).

I haven't looked at the fin split halfway up since I fully expect OR to get it wrong, and it's another case where the geometric reasoning necessary to get it right just doesn't seem reasonable.

teyrana, 2020-03-03:
> Do we want to remove the check, so we'll get the warning on 0-diameter pods?).

/endorse

I'm not sure why it was special-cased to begin with?

> I haven't looked at the fin split halfway up since I fully expect OR to get it wrong, and it's another case where the geometric reasoning necessary to get it right just doesn't seem reasonable.

/agree
I'm assuming you're saying the split-fin should simulate the same way as a one-piece fin?  Then yes, I agree.

neilweinstock, 2020-03-03:
Do we want to remove the check, so we'll get the warning on 0-diameter pods?).

/endorse

I'm not sure why it was special-cased to begin with?

Well, I don’t want a thick-fin warning when using 0-diameter body tubes.  That’s a bogus warning that I must stare at eternally.  I also don’t want the fin-body interaction to be calculated with 0-diameter bodies; I can’t tell from this discussion if that’s contributing any weirdness to the sim results.

Unless some bigger changes are made, phantom body tubes are *necessary* to implement fins-on-fins, which is a common design element and a useful capability of the pods feature.  It would be good to treat them as elegantly as possible, even if it’s impossible to make the underlying simulation perfectly accurate.

I haven't looked at the fin split halfway up since I fully expect OR to get it wrong, and it's another case where the geometric reasoning necessary to get it right just doesn't seem reasonable.

/agree
I'm assuming you're saying the split-fin should simulate the same way as a one-piece fin? Then yes, I agree.

I don’t expect that case to necessarily simulate correctly, but I wouldn’t have expected the results I’m getting.  I would appreciate if you’d take a look at it and make sure it’s doing something reasonable.

  *   Neil

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

JoePfeiffer, 2020-03-03:
It's just warning that the simulation is likely to be inaccurate, which is true, so it seems appropriate (just as we have similar warnings for things like supersonic flight).  There are certainly designs for which 0-diameter pods are needed, but it's likely those designs won't be simulated very accurately.

It does seem like the wording could be clearer; those fins aren't anybody's idea of thick and a user would have to be pretty familiar with OR to understand why a warning saying they are would turn up in this case.  Maybe a longer warning, "fins are thicker than 1/4 parent body tube diameter" or similar could clarify?

In the split fins, what I'm meaning to say is that ideally they would simulate the same as a one-piece fin, but I don't expect that to happen.  I'll try to get a chance to look at it some more.

wcravens, 2020-03-03:
I was thinking about this kind of thing the other day.  I cringe every time
I hear something like 'zero-diameter'.  The program should complain loudly
and deeply if a zero-diameter body tube (or that kind of thing) is ever
created.  Or a zero unit wall thickness.  I can just about imagine reasons
for having an object with zero mass, but in some ways that's a
stretch too.  The scientific computing nature of the program should utterly
reject these kind of 'imaginary' constructs and refuse to simulate at all.
But that's my opinion and I know that I'm pretty far along on one side of
that argument, which is why I don't pipe up more often.  But spring is in
the air, so today was a day to squawk.

All the best,

Wes

teyrana, 2020-03-03:
> Well, I don’t want a thick-fin warning when using 0-diameter body tubes.  

This is really the core of it -- that warning is intended to warn a user that they're not get accurate results.   The program isn't stopping anyone from doing what they want, it's just warning them that it won't be accurately simulated (perhaps annoyingly so) but it's not wrong.

> It does seem like the wording could be clearer; those fins aren't anybody's idea of thick and a user would have to be pretty familiar with OR to understand why a warning saying they are would turn up in this case. Maybe a longer warning, ... 

So, are  you're saying the error messages should be replaced?  Or reworded to be more descriptive?

teyrana, 2020-03-03:
In the split fins, what I'm meani>ng to say is that ideally they would simulate the same as a one-piece fin, but I don't expect that to happen.

_Ideally_, yes.  I'll see if there's money in the budget for it.... 😬 

> The scientific computing nature of the program should utterly
reject these kind of 'imaginary' constructs and refuse to simulate at all.

Yea, I do sympathize.   I would LOVE a hobby-grade CFD program that let us do things the right way.  Eventually, we might look into better ways of representing these more exotic designs.... but in the meantime, we're kinda stuck with this. :P

neilweinstock, 2020-03-03:
Wes,
I can understand your viewpoint but I disagree with it strongly.

OpenRocket is a versatile program with many uses.  If folks have found unexpected ways to get utility from the program that makes them happy, why would you want to take it away from them?

Phantom body tubes and other weirdo constructs are often necessary to make it possible to enter a design (designs which *can* be created in Rocksim).  If you don’t want them to be used, then make it so they’re not necessary.  For instance, the ability to mount fins to nose cones and transitions in the new version eliminates one situation where they were required in the past.  That is good.  However, even in the new version they are still necessary in order to implement fins-on-fins.  If the simulation is going to be inaccurate than just give me the warning (and as always, preferably one that I can understand) and let me decide how I want to continue.

I am a product developer by day. All the time we find out weird and unexpected ways customers are using our products.  Occasionally we actually flesh out features to make those use cases work better.  The last thing we would ever do is disable functionality that customers are using.

  *   Neil

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

JoePfeiffer, 2020-03-03:
To what extent do we want to let users create designs we can't simulate accurately?  And what do we want to do about it?

My answer is simulation isn't the only purpose of OR.  If it were we wouldn't have the 3D views, the material color selection...  People also use OR to get a sense of the aesthetics of a rocket, to create a BOM, and to create templates.  I don't think we should take those capabilities away just because the rocket can't be simulated well.

If the user does something that we can anticipate will screw up the simulation, we should let them know in as clear a way as possible.  The warning should be there (and it should be there with 0-diameter body tubes as well as .01 diameter body tubes), and it should say specifically that the fin thickness being greater than 1/4 the parent body tube diameter is why it's anticipating inaccuracies.

One note in this particular case is that body-fin interference (I wish it were called "interaction" but it's too late for that change) increases the effectiveness of the fins.  So putting the fins on 0-diameter pods moves the calculated CP forward -- while the simulation is in error, it's being conservative.  Though now that I say that I finally notice that it is indeed odd that the projected altitude is higher with the fins on pods.  Hmmmm

teyrana, 2020-03-03:
> Though now that I say that I finally notice that it is indeed odd that the projected altitude is higher with the fins on pods. 

Uh, yea. It's entirely possible that there is *also* a bug, here. :/

wcravens, 2020-03-04:
Just to try and un-dig my hole a little, I've never suggested stopping
people from doing their esoteric modelling strategies.  But I'd feel more
comfortable if it would just outright refuse to simulate if certain
constraints on the model aren't met (I know can open, worms everywhere).

JoePfeiffer, 2020-03-17:
No, I hadn't forgotten about this!  It turns out the drag calculations are still iterating through components and not instances (so one fin on each of three pods only has the drag of a single fin instead of three fins).  Working on it...

neilweinstock, 2020-03-17:
Thanks for following up on this, I thought something looked fishy.

JoePfeiffer, 2020-03-25:
Just submitted PR #603 addressing this.  Simulations with the two rockets that started this issue now agree to within a foot.  Jar file is at https://www.dropbox.com/s/rnzs2i0xp9e5l0z/OpenRocket-PR603.jar?dl=0 if anyone would like to give it a try.

neilweinstock, 2020-03-26:
Fantastic, thanks for sticking with this one.   Things are definitely looking much better.

Two questions:

  1.  CP is somewhat further forward on the pods versions.  Is this properly explainable?  I’m still unclear what causes the difference, at least on the second set of test files I provided where there is the exact same amount of fin/body interaction.
  2.  I’m seeing no thick fin warning on the pods versions.  Is that due to a decision or is that being handled in another issue?

Thanks,
Neil

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

JoePfeiffer, 2020-03-26:
>1.  CP is somewhat further forward on the pods versions.  Is this properly explainable?  I’m still unclear what causes the difference, at least on the second set of test files I provided where there is the exact same amount of fin/body interaction.

On the first pair, I was able to establish that the difference was perfectly accounted for by fin-body interaction.  Haven't looked at the second, but since span is one of the parameters on fin-body interaction I expect that's from the same cause.

> 2.  I’m seeing no thick fin warning on the pods versions.  Is that due to a decision or is that being handled in another issue?

That's issue #571, which we haven't decided how to resolve yet. I was thinking Dan had submitted a PR to try to move us along on this, but I'm not seeing it in a brief search.

neilweinstock, 2020-03-26:
1.       CP is somewhat further forward on the pods versions. Is this properly explainable? I’m still unclear what causes the difference, at least on the second set of test files I provided where there is the exact same amount of fin/body interaction.

On the first pair, I was able to establish that the difference was perfectly accounted for by fin-body interaction. Haven't looked at the second, but since span is one of the parameters on fin-body interaction I expect that's from the same cause.

The second is the more interesting example.  I’ll keep testing and see if I can understand and characterize the new behavior better.  I’m not yet convinced that it’s perfect (it might be, just not sure yet), but it’s definitely getting there.

  *   Neil

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

JoePfeiffer, 2020-03-26:
It is counterintuitive that when you split a fin axially (as in your second pair of examples) you get a different CP and weight than with the original fin, but that's what comes out of equation 3.40 in Sampo's thesis (which is where we get the formula for CNA1).

Taking the simpler case of a square fin with span 1, Aref=2pi, beta=1, when I plug the numbers in I get CNA1=1/(1+sqrt(2)). If I split the fin axially so I've got two fins with span 1/2 and area 1/2, the sum I get for the two of them is 1/(2+sqrt(5)).  So yes, it  should be different.

neilweinstock, 2020-03-26:
I accept your explanation.  Thanks for walking me through it.

Here is an important but completely unrelated problem I’m seeing with this build, which probably has nothing to do with a change to CP calculation: periodically it will consume a lot of CPU just sitting there doing nothing:
[cid:image001.png@01D6037C.2B564370]

There it is taking 13.9% of my 6-core i7, which is a lot, and enough to get my fans going.  It stays like that for a long time (maybe indefinitely, but I haven’t waited long enough to see).  It also doesn’t *always* do it, but every time I’ve started it up and opened a file I’ve been able to get it to do this eventually.

I have not been able to reproduce this with 15.03.  Haven’t tried it with any of my other dev builds yet.

Do you want to look at it here, or should I file separately?

  *   Neil

JoePfeiffer, 2020-03-26:
I'd say file it separately.  It almost has to be something else.

neilweinstock, 2020-03-26:
OK.  In that case you can close this one.

JoePfeiffer, 2020-03-26:
Let's wait for Dan to review and (hopefully) accept the PR before we close.

PR #603 (2020-03-29): [Fixes 568] Consider all instances of components when calculating drag
Fixes #568 by iterating through all instances of rocket components when calculating drag.
Adds unit test for fins on pods as in bug report.

Jar file is at https://www.dropbox.com/s/rnzs2i0xp9e5l0z/OpenRocket-PR603.jar?dl=0 if anyone would like to take it for a spin.
closed 2020-03-29 by teyrana