openrocket/openrocket #571: Add more descriptive messages for Fins-on-phantom-bodies

teyrana, 2020-03-03:
Currently the warning is something like "thick fins -- accuracy may suffer".  Change this to something like "fins on phantom bodies are allowed, but discouraged -- simulation accuracy may suffer"

Comments on wording are welcome!

@neilweinstock @JoePfeiffer

wolsen, 2020-03-04:
What about: "fins attached to phantom body tubes may produce inaccurate simulations"

neilweinstock, 2020-03-04:
Here are the three warnings I know about (there may be others):
[image]

You'll note that they are (a) not consistent, and (b) repetitive (that is, they basically say "may not be accurately modeled" twice, and we're talking about adding it a third time.)

Ideally, at some time in the future, the structure and presentation of the warnings can be improved.  In the meantime, I'd try to make the new warning as consistent as possible with (one of) the old ones.  How about: "Fins attached to phantom body tubes may be modeled inaccurately".

- Neil

JoePfeiffer, 2020-03-04:
There are a bunch of possible warnings that appear at different points in the design/simulation cycle, see https://github.com/openrocket/openrocket/blob/unstable/core/resources/l10n/messages.properties#L1685

In the case of this particular warning, we want to better capture that the warning relates to the relative thickness of the fins and the body tube they are attached to -- I don't think we want a separate warning for the case in which it's a phantom body tube, and I'd like to see it be clear as to what the exact criterion for the warning is.  So I like something more like "Simulation results may not be accurate for fins thicker than 1/4 parent body tube diameter"

neilweinstock, 2020-03-04:
There are a bunch of possible warnings that appear at different points in the design/simulation cycle, see https://github.com/openrocket/openrocket/blob/unstable/core/resources/l10n/messages.properties#L1685<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fopenrocket%2Fopenrocket%2Fblob%2Funstable%2Fcore%2Fresources%2Fl10n%2Fmessages.properties%23L1685&data=02%7C01%7Cneil.weinstock%40raritan.com%7Cf89b5e3f4cfb441f62fc08d7c04dadba%7C199686b5bef4496087867a6b1888fee3%7C1%7C0%7C637189312167014679&sdata=CQcUfSunfPZwpGMIIiNY2kcwvuQ%2BVR32VeQA4nJScAg%3D&reserved=0>

Ah, I forgot “Too many parallel fins”, which doesn’t make any mention of what the actual problem is (i.e., inaccurate modeling).  As I said, the messaging is inconsistent.  Also note that much like the *current* thick fins warning, the parallel fins warning doesn’t say how many is too many.

Best case implementation would be to have a brief warning displayed in the view area (e.g.: “too many parallel fins”, “thick fins”, etc.), with a more detailed tooltip containing a full explanation (or a link to an on-line help document).  I don’t expect to get there this release, but it would be a worthy goal for the future.

In the case of this particular warning, we want to better capture that the warning relates to the relative thickness of the fins and the body tube they are attached to -- I don't think we want a separate warning for the case in which it's a phantom body tube, and I'd like to see it be clear as to what the exact criterion for the warning is. So I like something more like "Simulation results may not be accurate for fins thicker than 1/4 parent body tube diameter"

A special warning for this case is simply clearer for the user and has no downside that I know of.  In my mind a phantom body tube is not a body tube at all, so that message requires a bit of brain processing to get what it’s saying.

There have been a good number of support questions on TRF where the user does not understand the meaning or source of a warning (top two examples: “body diameter discontinuity” and “high speed deployment”). Any time we can make a warning or error message clearer then we should, especially in cases such as this where the warning-generating code has sufficient visibility into the issue to make the distinction without a bunch of extra fiddly work.

This is not a hill I want to die on for this release, but it’s the way I view this issue.

  *   Neil

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2020-03-22:
For Future Reference:
-----
Call Site: https://github.com/openrocket/openrocket/blob/af3d226b2b53aafa28a4b57181dd18167c414372/core/src/net/sf/openrocket/aerodynamics/barrowman/FinSetCalc.java#L105

teyrana, 2020-05-17:
Addressed in #670.  Take a look if interested.

commit 289e0cb4 (2020-05-22): Merge pull request #670 from teyrana/571-fin-phantom-body-message

[Resolve #571] Add message fins-on-phantom-bodies
closed 2020-05-22 by JoePfeiffer