openrocket/openrocket #573: Errors opening photo studio

wolsen, 2020-03-05:
Giving the latest commit for a spin, which adds Java 11/13 support and switches 3D rendering to JOGL, the Photo Studio fails to render when it tries to paint with the following exception:

1340     46.132 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.NullPointerException
1341     46.132 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.gui.figure3d.photo.sky.SkyBox.square(SkyBox.java:95)
...

Looking at the log more and the code, it appears to be due to failing to load textures.

1305     46.033 DEBUG  (TextureCache.java:84) Loading texture file:/home/wolsen/git/openrocket/swing/resources/datafiles/sky/box/North.jpg
1306     46.119 ERROR  (TextureCache.java:89) Error loading Texture
   > java.lang.IndexOutOfBoundsException: Required 1048576 remaining bytes in buffer, only had 786432

Researching around, this error occurs when the texture data being loaded is expecting a different size/format of the data than what it is expecting.

wolsen, 2020-03-05:
It appears that the IndexOutOfBoundsException can be corrected by specifying GL.GL_RGB instead of GL.GL_RGBA when creating new TextureData in the TextureCache. It resolves the loading of the texture, but results in an upside down texture.

[image]

I think this can be resolved by creating the texture using AWTTextureIO instead.

wolsen, 2020-03-05:
After switching to use the AWTTextureIO everything looks good in the Photo Studio. This is using the linked branch.

[image]

wolsen, 2020-03-05:
The upside down image can also be seen when loading the example rocket 'Boosted Dart'.

teyrana, 2020-03-08:
Confirmed: Reproduced 2020/03/08.

Teals53, 2020-03-09:
@wolsen openrocket have simulation animation like that ? or how can i use that i really wonder

Teals53, 2020-03-09:
anyone can explain me  that ? i really excited dude can u tell me what happening there ? how can i use last openrocket build ? when i try to build that its have a lot of bugs etc and its not useful. now last openrocket build have simulation like that or i need use other software for make that ? can you help me asap ? thank you @wolsen  @teyrana

wolsen, 2020-03-10:
@Teals53 OpenRocket does not have a simulation animation. The photos above are from the Photo Studio, where OpenRocket can show a 3d rendering of your rocket against a few different backgrounds.

If you're having problems using the unstable build, you should probably consider running from a stable version for the time being (15.03). Yes, its a bit older and doesn't have some of the newer features (pods, boosters, etc), but it does have most things you need for designing simple to complex model rockets. The user's guide is available in wiki form (though it is being migrated to github wikis) @ http://wiki.openrocket.info/User%27s_Guide and there are several tutorials available online via YouTube and others.

Hope that helps get you started.

Teals53, 2020-03-10:
> @Teals53 OpenRocket does not have a simulation animation. The photos above are from the Photo Studio, where OpenRocket can show a 3d rendering of your rocket against a few different backgrounds.
>
> If [...]

Ohh thank you for help also i want to learn how can i build newest unstable version truly ? Without broken things

teyrana, 2020-03-11:
Hey Bekir,

If you're interested in building and helping with the latest unstable
version, we can always use more help!  Where to start? Well that depends on
how much programing you know ... Can you tell us a little bit about your
experience level?

> I want to learn how can i build newest

 > unstable version truly ? Without broken things

Well the problem us, that the unstable branch, by definition, has broken
things. If you *don't* want "broken things" then you want the official
releases (see previous emails)

The problem us that "Unstable" is, by definition, a repair shop where we
figure that stuff out.  It may crash, it may write files that can't be read
anywhere--sometimes, even by itself...  It may remove a feature you've been
using... and finally, it may not even run on your computer.

But, if you don't mind the mess, and want to help us fix all these things,
we'd love to have you :)
- Dan

Teals53, 2020-03-11:
Actually i dont have enought code skill for developing this software but when i build last build its have weird something for example some options working in released version but not working on the latest build and i think its depends my packing style and i am still not sure its caused by me or just its have broken source can you send me latest build compiled from you ?

PR #574 (2020-03-11): Use AWTTextureIO for creating textures from jpegs
The switch to using JOGL uses the current JPEGs, but these textures
cannot be read using the GL.GL_RGBA format which is eventually
propigated to the user by an error dialog showing a NullPointerException
for the texture which failed to load. Using GL.GL_RGB as the format when
loading the textures will resolve the texture loading issue, but the
images are inverted due to differences between AWT and JOGL
interpretation of where the image starts (top left vs bottom left).

Change instead to use the AWTTextureIO class to create the textures
which s [... cut]
commit 57b92857 (2020-03-11): Use AWTTextureIO for creating textures from jpegs

The switch to using JOGL uses the current JPEGs, but these textures
cannot be read using the GL.GL_RGBA format which is eventually
propigated to the user by an error dialog showing a NullPointerException
for the texture which failed to load. Using GL.GL_RGB as the format when
loading the textures will resolve the texture loading issue, but the
images are inverted due to differences between AWT and JOGL
interpretation of where the image starts (top left vs bottom left).

Change instead to use the AWTTextureIO class to create the textures
which  [... cut]
closed 2020-03-11 by teyrana