openrocket/openrocket #590: Runtime Error with Java 13

acxz, 2020-03-13:
I tried to build the latest default branch (unstable) and got it working with Java 13. (`ant jar` after cloning.)
However, when I tried to run OpenRocket like so: `java -jar swing/build/jar/OpenRocket.jar`
I got the following error:
```
acxz@archard ~/vcs/git/github/openrocket/openrocket (git)-[unstable] % java -jar swing/build/jar/OpenRocket.jar 
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/home/acxz/vcs/git/github/openrocket/openrocket/swing/build/jar/OpenRocket.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!
java -jar swing/build/jar/OpenRocket.jar  6.81s user 0.44s system 204% cpu 3.556 total
```

wolsen, 2020-03-13:
Which jdk are you using? This looks like a bug identified in the openjdk 11..13 version with Ubuntu/Debian - https://bugs.launchpad.net/ubuntu/+source/openjdk-lts/+bug/1838740.

A known work around is to use the adoptopenjdk version, though the Ubuntu/Debian java runtimes should be fixed.

acxz, 2020-03-13:
I am using openjdk13 on ArchLinux.
Let me give adoptopenjdk a shot.

acxz, 2020-03-13:
bah dum, adoptopenjdk worked! Thx so much, since this bug is not related to openrocket I'll go ahead and close the issue.

closed 2020-03-13 by acxz