openrocket/openrocket #592: Example Rockets have loading errors

teyrana, 2020-03-16:
## Example 1: "Apocalypse with Decals"
- Error: "No Motor with designation 'J90W' for manufacturer 'AeroTech' found."
- Motor exists on Thrustcurve.org as " AeroTech J90"
- "example" .ork file saves this motor as `<designation>J90W</designation>`
- "fixed" .ork file saves this motor as `<designation>J90</designation>`

teyrana, 2020-03-16:
@JoePfeiffer 
Hey, I've traced this change back to
https://github.com/openrocket/openrocket/commit/6c535bff94d5ad4fbc06c86c12d096d1fcb0a99b#diff-91e4e7242e67530a1198a37c9911a7f8

Do you happen to remember the context for this? 

Was this motor removed, or did we just change how we indexed them? 
(I can update the example rocket, but I need to know which option we're looking at....)

teyrana, 2020-03-16:
FYI, this is my fixed version: All I've changed is the missing motor.
[fixed-example-apocalypse.ork.zip](https://github.com/openrocket/openrocket/files/4340705/fixed-example-apocalypse.ork.zip)

JoePfeiffer, 2020-03-17:
It stems from Issue #516, where several motors were found to crash the simulation.  The problem turned out to be those motors had errors in their thrust curves on thrustcurve.org; the fix  was to find good heuristics to fix the curves we could and not enter the ones we couldn't in the database (John Coker was very hesitant about losing the data from removing curves from his database, and preferred to keep what data he had).  I'm going to have to take a look back at the changes; some of the motors still have suffixes while others (like this one; J90 instead of J90W) don't in our database.

teyrana, 2020-03-19:
Oh, I remember that one.

So were those motors *removed*? or fixed?

Basically, is "J90W" the same as a "J90" for example purposes?

teyrana, 2020-03-19:
I did some more research -- I "fixed" the file.... but when you load it through examples, it throws an error, but when you load it "normally" it can't find that very same file. 

It looks like a loading error.

JoePfeiffer, 2020-03-20:
30 motors were removed; I didn't log the ones that were fixed. I think all of the suffixes got removed (so J90, not J90W) but I don't remember why.  Now I'm curious about that...

closed 2020-03-22 by teyrana

JoePfeiffer, 2020-04-02:
Spent some time tracking down exactly what happened here, for future reference.  Motors are saved with rocket configurations in two forms:  a designation (like J90) and a digest, which amounts to a hash of the motor characteristics.  Ordinarily, when a rocket's configurations are loaded the digest is used to look up the motors.  If there is no motor with a matching digest, the motor designation is used.
In 15.03, the designation is the actual manufacturer's motor designation, as in the J90W that sparked this or one of those long, long CTI designations.  At some point, this was switched to using a motor's common name like  J90.
As long as a motor's digest hasn't changed, loading the configurations will happily find the correct motor in spite of the designation having changed.  On the other hand, if the digest has changed but the designation has not, it will find the correct motor and put up a warning that the motor's thrust curve has changed.
What happened in this case is the designation had changed (from J90W to J90) previously, and the change in thrustcurve handling resulted in a change to the digest as well.  As a result, it wasn't able to find the motor.

teyrana, 2020-04-04:
Yea, I figured it was something like that.  So are we happy with the current behavior? u

JoePfeiffer, 2020-04-04:
Would it be worth putting up a warning when a motor designation doesn't match its thrust curve, or maybe do a fallback search on the manufacturer's designation (and put up a warning if found)?

teyrana, 2020-04-04:
... that's what I'm asking you.

commit af3d226b (2020-03-22): [fix #592] Fixes the 'Apocalypse with decals' example rocket