openrocket/openrocket #600: Calculates Incorrect Cp on split-fin rockets

teyrana, 2020-03-22:
### Reproduction: 

Open the following in current development branch (`unstable`):
[simple_split_fins.ork.zip](https://github.com/openrocket/openrocket/files/4365711/simple_split_fins.ork.zip)

### Expected 
Same as in 15.03, or the equivalent 4-fin FinSet
Cp.x = 50.5 cm 

### Actual
Cp.x = -32.7 cm

JoePfeiffer, 2020-03-22:
-32.7?  Wow.  I'm still plugging through the BaroowmanCalculator which includes both that code and the friction code; when I get the PR finished for the friction code I can look at this if you'd like.

teyrana, 2020-03-23:
@JoePfeiffer 

Actually I have a PR that you can look at, if you have the time /grin
(#601)

I've definitely fixed this *particular* issue (#600) but I'm not sure that the rest of the associated code works the way we want.

PR #601 (2020-04-06): [Request Review] Corrects the cp calculations for Split-fin rockets.  Fixes #600.
### Inciting Example
[simple_split_fins.ork.zip](https://github.com/openrocket/openrocket/files/4365711/simple_split_fins.ork.zip)

### Relevant Unit Tests:

https://github.com/teyrana/openrocket/blob/2c1daea9d1eea139eb5d55e6ac397e4ec67ef61d/core/test/net/sf/openrocket/aerodynamics/BarrowmanCalculatorTest.java#L114-L226

(prebuilt jar available on request)

Fixes:  #600
May help with: #571, #538
commit 58eb9fff (2020-04-06): Merge pull request #601 from teyrana/fix/600/split-fins-cp

[Request Review] Corrects the cp calculations for Split-fin rockets.  Fixes #600.
closed 2020-04-06 by teyrana