openrocket/openrocket #604: 3D Print Export Feature

loic-fejoz, 2020-03-28:
It would be nice to export the design to a CAD / 3D modeller software so as to be able to 3D print (or eventually cast) some parts.

I have an [on-going code repository for IceSL export](https://github.com/loic-fejoz/openrocket/tree/icesl-export).

For instance, one could 3D print the ogive and a lower tube+fins:

[image]

NB: Eventually another export to STL file format could be nice too.

JoePfeiffer, 2020-03-28:
This is something I'd seriously love to see.  Too much existing stuff to get working again for me to work on this one, though.

loic-fejoz, 2020-03-28:
@JoePfeiffer, as I said, I already start to work on this issue. I already have a working code exporting to IceSL. All ogive shapes are already properly generated. I still have to look at fin shapes more deeply (chord and stuff like this, but overall trapezoid shape is managed). Once done with IceSL, I think I would be able to do it for STL natively.
So should I submit already a pull request in order to discuss implementation more in details? Or should I wait before having a more solid implementation?

teyrana, 2020-03-28:
Right now, we're trying to get a new release out.

That mean fixing a lot of bugs.   While I do like the idea of exporting stl
files, adding it to this release would bog down everything so much, it'd be
hard to get the thing released anytime soon.

Would love to see it in a subsequent release, though !  :)

Daniel Williams
equipoise@gmail.com
858-220-8134 (Mobile)
github.com/teyrana
<https://github.com/teyrana>LinkedIn
<https://www.linkedin.com/in/williamsdm>

loic-fejoz, 2020-03-29:
@teyrana  No requirement from my side for this feature to be in this particular release. It was more make people aware on the on-going work. I still have one question about the airfoil profile of fins. I guess it is expected to be symmetrical air-foil but which NACA number is it considered?

teyrana, 2020-03-29:
ahhh, I'm not aware that it was ever that specific.

I would guess most of us just round off the fins, and call it a day.
Also, I'd be curious how many of us have access to a 3d printer with
sufficient precision for a thinner 0009 or 0006.  IMHO, I'd just default to
the 0012, I think.

Maybe it's a feature you can add ?
Or add to V2.0?

Daniel Williams
equipoise@gmail.com
858-220-8134 (Mobile)
github.com/teyrana
<https://github.com/teyrana>LinkedIn
<https://www.linkedin.com/in/williamsdm>

JoePfeiffer, 2020-03-29:
All of the models OR is based on assume a flat plate fin, so OR does too.  For 3D printing, adding a profile would be a nice feature, but it wouldn't affect the simulations.

Which 00xx airfoil is used in a particular model would just be thickness/chord as a percent, and of course for a tapered fin that would mean the thickness would vary across the span of the fin.

wolsen, 2020-03-29:
Disclaimer: I don't have a 3d printer nor 3d printing experience.

That being said, I love the idea of an export which allows for 3d printing. It seems its a common feature request popping up for rocketry software. I think there's a few changes necessary in OpenRocket to get some of the bits right (like fin profiles, etc). I also personally lean towards favoring an STL implementation first as it is more generic and would allow existing users to use 3D printing in the way that they are most comfortable. This also allows it to benefit a larger user base.

I also think this would be a great feature for the next release.

loic-fejoz, 2020-03-30:
@JoePfeiffer Apparently the simulator already manage airfoil: indeed on the simple model rocket, modifying a flat fin to an airfoil one make apogee gain a few meter...

site note:
For all, that have no access to 3D printer, you may be interested in looking at the FabLab maps:
https://www.fablabs.io/labs/map

@wolsen I agree with you on all point. IceSL was just a starting point because of its high level feature. Once I understand all internals of OpenRocket, I will be able to do it for STL (I think I can already in fact).

BUT STL is less flexible for multiple parts generation. So maybe, this feature would be better suited if we introduce a contextual menu on the parts tree.

Another alternatives would be to generate a standard CAD file (let's say a STEP file).

@teyrana Anyway I don't think it can make it for the next release.

JoePfeiffer, 2020-03-30:
All it's doing is adjusting the fin pressure drag depending on whether the leading and trailing edges are square, rounded, or (for the trailing edge) tapered.  No serious airfoil modeling.  What's happening is documented in section 3.4.4 of the technical documentation.

loic-fejoz, 2020-06-29:
Small update on this ticket, export of nose cone is done and integrated in the GUI as such:
[image]

I think a component per component export is more adapted.

Where do you the button should go?

wolsen, 2020-06-30:
Hey that's cool.

I feel that this should probably fit within the current export flows. There's an export option in the file menu which exports to RockSim. I think it would probably flow better if the File menu had an export sub menu which allowed the user to select RockSim, STL, etc... That would likely need some UI thought as you'd need to show a dialog which allows the user to select which components they want to export. I can see a user designing a rocket and then wanting to export a fin can and some nose cones as part of this. There's some precedence for this in the print/export pdf dialog.

wolsen, 2020-06-30:
As a thought, its probably worth putting some ideas down in some mockups & specs so we know when it gets implemented how it should work vs how the implementation is. Gives a place to better chew on some of this stuff.

PR #2301 (2023-09-05): [#604] Implement 3D component exporting to Wavefront OBJ
This PR fixes #604 and introduces the ability for users to export rocket components as Wavefront OBJ 3D objects.

### Why OBJ Format?

- **Editability**: OBJ files are easily editable in 3D software (e.g. Blender), compared to e.g. an STL file. Granted, it's less easy to use in CAS software than STEP-files, but STEP-files are difficult to generate without mesh errors.
- **3D Printing Compatibility**: Most 3D printing slicers directly support OBJ importing. If not, there are a ton of free and easy-to-use converters available to export OBJ [... cut]
closed 2023-09-05 by SiboVG