openrocket/openrocket #609: Warnings on build

teyrana, 2020-03-29:
## The warning message
```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/D:/project/openrocket/build/jar/OpenRocket.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

## behavior changes
none-known. yet.

## Suspected location
guice libraries:  note the `com.google.inject` namespace.

itshanney, 2020-04-02:
https://github.com/google/guice/issues/1133

This is from the switch to Java 9+ and the adoption of the module system that's inherent since Java 9. The addition of the module system changes the classpath from the way it was handled before Java 9.

`--add-opens java.base/java.lang=ALL-UNNAMED`

Adding the above as an argument when you run the program removes the warning, but it is a nuclear option, as mentioned by the guice issue. Permanent fix is to ensure Guice usage is up-to-date with Java 9+, which we will cover as part of this issue.

teyrana, 2020-04-07:
fixed by #614

closed 2020-04-07 by teyrana