openrocket/openrocket #613: In recent unstable build, app name shows as "Swing Startup"

neilweinstock, 2020-04-01:
While testing a recent build on openJDK11 on Mojave, the app name showed in the menu bar as "Swing Startup".  Operation seemed to be normal otherwise.

[image]

itshanney, 2020-04-02:
@neilweinstock how are you starting/launching this? In your IDE, if you run the main() method of SwingStartup, it will start and show are you present. You need to be running the main() method from OpenRocket.java instead.

I just ran this from my IDE and it works as expected as well as running the .jar built by ant. Can you check to make sure you're running OpenRocket::main()?

neilweinstock, 2020-04-02:
I’m running a jar file via command line: “java -jar <jarfile.jar>”.  Shouldn’t that work?

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

itshanney, 2020-04-02:
@neilweinstock yes, that should work. I'm just not seeing the same symptoms you are seeing. I just pulled down latest unstable and ran it and it works as expected from both the jar file and when I run in my IDE. 

Can you send me the .jar file that's broken?

itshanney, 2020-04-06:
@neilweinstock are you OK with calling this a fluke build and the issue is resolved?

neilweinstock, 2020-04-07:
Sure, can reopen if the problem reappears.

itshanney, 2020-04-07:
Problem appears to be from a spurious build that's since resolved itself. Closing for now and can reopen if it presents itself again.

closed 2020-04-07 by itshanney