openrocket/openrocket #615: Large CPU usage in unstable branch

itshanney, 2020-04-03:
When running the latest build from unstable, there is a very high bit of CPU usage. I hooked up a profiler to try and see what the culprit is and the attached screenshots show the top consumers of CPU. 

[image]
[image]
[image]

itshanney, 2020-04-03:
Found the culprit. It's from this chunk of code starting on line 364 in ScaleScrollPane.java.  It's continually drawing the ticks and calling the drawTick method.

```java
for (Tick t : ticks) {
    int position = toPx(t.value);
    drawTick(g2, position, t);
}
```

teyrana, 2020-04-04:
well it's running within the outer 'paintComponent` method...  which is intended to only be called when something changes.

wolsen, 2020-04-04:
@itshanney Ah good find. Looking at the code, I think that the ScaleScrollPane.Ruler.paintComponent() method is causing this. In that code, one of the first things it does is invoke updateSize() which in turn calls revalidate() and repaint(). AIUI, this will cause the underlying graphics libraries mark the component as dirty. The paintComponent method will continue to execute, but the component is marked as dirty which will then retrigger an invocation to the paintComponent... which in turn calls updateSize() and we are then on the merry-go-round of painting those rulers super frequently.

wolsen, 2020-04-04:
Here's a [test build](https://www.dropbox.com/s/im0no05u8ytlpil/OpenRocket-fix-615-high-cpu.jar?dl=0) with the changes included for anyone who wants to test. See if this resolves the high cpu usage or if we need to chase some further activity. It looks promising on my machine.

@itshanney @neilweinstock - can you guys give this a go?

neilweinstock, 2020-04-05:
Thanks, I'll give it a test tomorrow and get back to you.

neilweinstock, 2020-04-05:
Looks good on MacOS 10.14.

  *   Neil

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

PR #616 (2020-04-05): Don't revalidate/repaint component in Ruler.updateSize()
Analysis from Justin Hanney indicates that the CPU is constantly
processing paint events in the ScaleScrollPane.Ruler. Change the
updateSize method to not revalidate and repaint since the
paintComponent method also invokes the updateSize to ensure it is
drawing components of the right size.

This change also refactors the ChangeListener implementation to
have the RulerClass implement the method instead of using an
anonymous class for event handling when ruler units change.

Additionally, this change adds a revalidate call to the scrollPa [... cut]
commit adf31013 (2020-04-05): Don't revalidate/repaint component in Ruler.updateSize()

Analysis from Justin Hanney indicates that the CPU is constantly
processing paint events in the ScaleScrollPane.Ruler. Change the
updateSize method to not revalidate and repaint since the
paintComponent method also invokes the updateSize to ensure it is
drawing components of the right size.

This change also refactors the ChangeListener implementation to
have the RulerClass implement the method instead of using an
anonymous class for event handling when ruler units change.

Additionally, this change adds a revalidate call to the scrollP [... cut]
closed 2020-04-05 by teyrana