openrocket/openrocket #618: Problem with display of fins attached to transition or nose cone

neilweinstock, 2020-04-07:
I am working with the build that Justin recently distributed in package form.

I experienced the following weirdness.  The attached design is stripped down to the bare essentials to show the bug.

I have a set of freeform fins that were originally attached to the body tube.  All was well:
[image]

I then moved the fins to the tail cone.  Invisible fins!
[image]

It also had problems display the root of the fin in side view:
[image]

Moving the fins to the nose cone behaves the same as on the tail cone.

Note that if I create *new* fins, they work just fine.  Just this pre-existing set of fins is having trouble.  It could be related to the fact that the *original* version of these fins had a "cutout in them" which may have created a backward loop at one point.  But I've deleted all those extra points and now they're simple four-point fins, and they persist in displaying this way.  Hopefully you can see what's wrong in the ORK file.

[clear-fins.ork.zip](https://github.com/openrocket/openrocket/files/4441590/clear-fins.ork.zip)

teyrana, 2020-04-07:
:/   That's not a "cutout".  That's the fin tab.

Basically, when I implemented the feature, I picked the first tactic that came to mind.  I think the tab depth/width is currently measured from it's midpoint, iirc?  

My question is -- did you *intend* to configure the tabs?  (did you edit them at all?)

If so, how do you expect fin tabs to behave in this situtaion?

neilweinstock, 2020-04-07:
No, I indeed meant cutout, like this:
[cid:550fd0bc-99fb-4eec-acc4-448c7a4c7497]

I had fins go invisible once before when I had a an interior cutout, but those (if I recall correctly) came back to visible when I got rid of the cutout.  This particular set of fins is stuck in an alternate dimension.

You're right that the weirdness being displayed at the root is due to the fin tab, which somehow got set to .09".  Unclear how that happened, I sure don't remember doing it.  But it appears that we can forget about side view problem, which isn't a problem, and focus on the invisible fins.

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2020-04-07:
Yea, as you move the fin, there is logic, which attempts to adjust the fin tab....  

So, obviously there's a bug in it... But I could imagine that not being cautious enough.

I'll have to see if I can reproduce it, and what the steps are.

neilweinstock, 2020-04-07:
The weird thing is that I can move the fin back and forth between the transition and body, and it will alternate being visible (normal) and invisible…  Newly created fins do not have this problem.

teyrana, 2020-04-18:
##  Fixing the render behavior
Okay, I've made a patch for the render behavior here.  I think this makes more sense than the render we get above.  Let me know what you think.
https://github.com/openrocket/openrocket/pull/625

## I couldn't actually reproduce the bug you're asking about... can you try something for me? 
1. set the tab height to zero for the fin, in your save file.
2. then try to reproduce the behavior
   (and write down the steps :D)

Hopefully, this in one of those things which has been subsequently fixed ...

teyrana, 2020-04-26:
@neilweinstock 
This issue is now in mainline.  Were you able to verify if this behavior is resolved?

neilweinstock, 2020-04-27:
The side-view display of tabs is fixed.

The clear fins are still there.

teyrana, 2020-04-27:
Okay, I need more information about the clear fins, then... I'm not even sure what you're referring to.

Which screen would I find these on? 
And can you try uploading that picture again? It didn't take, the first time.

neilweinstock, 2020-05-01:
It's the second picture right up there in the first entry.  You can see the outlines of the fins, but not the surface, which is completely invisible.  It's the same image that shows the weird fin tab.  Do you not see it?

wolsen, 2020-05-01:
Ah, it took me a moment. It's where the 3d fins show up as white when transitioned backwards.

[image]

neilweinstock, 2020-05-01:
Yup, that's it.  I apologize I thought it was more obvious than it apparently is. :)

teyrana, 2020-05-02:
oh, okay, _now_ I see what you mean.   . . . and I can reproduce it.   

That IS weird!   It's not even "all" freeform fins, it's just that specific magic fin, when copied, or moved.

If i add a new fffin, it functions as desired.  

This IS funky.

neilweinstock, 2020-05-02:
Yes, it is very weird indeed.

My initial thought was that since these fins initially had that “cutout” in them, which was a cycle in the coordinates, then maybe that caused the coordinates to somehow end up in reverse order that that causes the transparent rendering.  It is a stretch, but I’m at a loss for any other reason.  In any case, the problem seems to stick with that particular finset.

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2020-05-16:
Fixed by #663 ;
Closing as resolved.

closed 2020-05-16 by teyrana