openrocket/openrocket #631: Travis no longer appears to be commenting on PRs

wolsen, 2020-04-20:
It seems recently the travis-ci has stopped commenting on bugs. It feels like it occurred around the java version switch, but I have no evidence of that.

itshanney, 2020-04-20:
@wolsen looks like your latest merge broke the build with failing unit tests:

Commit: https://travis-ci.org/github/openrocket/openrocket/builds/677291104
Build History: https://travis-ci.org/github/openrocket/openrocket/builds

History looks solid going back to the JDK11 upgrade.

itshanney, 2020-04-20:
@wolsen seeing the unit test failure when I run locally with latest code from unstable. Test report attached below with error in FreeformFinSetTest.

[test-report.tar.gz](https://github.com/openrocket/openrocket/files/4506053/test-report.tar.gz)

wolsen, 2020-04-20:
@itshanney ah yeah, see this is why we need travis commenting :-)

I'm not sure why its not commenting but I'll look at that unit test.

wolsen, 2020-04-21:
@itshanney, @teyrana - unit test fixed in PR#632

itshanney, 2020-04-22:
@teyrana did you or anybody else by chance remove the Travis Status Check for the repo when we were converting to Java 9+?

https://help.github.com/en/github/administering-a-repository/enabling-required-status-checks

The Required Status Checks appear to be what's forcing Travis to comment on pull requests. The first example below is "old enough" that it has the Status check and Travis is reporting. The second one is "too new" that it lacks the check and Travis is not reporting.

https://github.com/openrocket/openrocket/pull/559
https://github.com/openrocket/openrocket/pull/621

Can you take a second look at the Admin settings for the repo, or temporarily grant access to myself or @wolsen and we can take a look to turn it back on?

teyrana, 2020-04-23:
@itshanney 
_Not on purpose_!
(but I may have screwed it up, when we were porting from java 8 -> 9/11/13 ...)

teyrana, 2020-04-23:
I also changed the settings from the Ubuntu 14.04 image to 18.04.  It's not unlikely that contributed.

teyrana, 2020-04-23:
If we look at all the PRs, it looks like something around #599 (tested) to #601 (not tested)

https://github.com/openrocket/openrocket/pulls?q=is%3Apr+is%3Aclosed

teyrana, 2020-04-23:
The last pr to change `.travis.yml`:
https://github.com/openrocket/openrocket/pull/588/commits/8902a925f3f30fcf412bbebfa512633772691a36

teyrana, 2020-04-23:
BTW, I don't have enough permissions to re-enable things, I don't think.   Do we know what the problem is, yet?

teyrana, 2020-04-23:
@kruland2607 @wcravens 
BTW, I don't think I have permissions to re-enable Travis CI runs. Can one of you take a look?

Do we know what the problem is, yet?

closed 2020-04-25 by wolsen