openrocket/openrocket #637: Printing is slow and appears to hang

wolsen, 2020-04-27:
Printing a rocket is very slow without any user feedback to indicate what's going on. When a rocket has multiple flight configurations or long running simulations there is a long delay between the time the user clicks the "Preview and Print" or "Save as PDF" buttons.

**Reproducer Steps**
1. Open the "Boosted Dart" example rocket.
2. Select Print / Export PDF ... from the file menu
3. Ensure Design Report is selected and click "Preview & Print" or "Save as PDF"

wolsen, 2020-04-27:
The cause of this is largely due to the fact that the DesignReport will run each of the simulations that match a given motor id, regardless of whether the Simulation is up to date or not. This is done without using the SimulationRunDialog so there's no feedback provided to the user. To make matters worse, the Simulations are copied instead of run natively so the user pays the time to run the simulations but does not benefit from having the simulation data be up to date in the rest of the UI.

To fix this, I propose one of two options:

**Option 1**
Add a new option to the Print Dialog that allows the user to select whether or not they wish to update simulation data before printing. This will give them the following behaviors:

1. **User does not choose to update simulation data** the DesignReport simply uses the data that is present.
2. **User does choose to update simulation data** the DesignReport will run only those simulations that are not up to date and any simulations run will be run using the SimulationRunDialog to offer feedback to the user. Additionally, the simulations will be run on the original simulations, not a copy, in order to have the simulations updated in the UI and OpenRocketDocument.

**Option 2**
Simply use the simulation data is present and skip any running of simulations. This is by far a simpler solution and nets the same benefit, however it is a changing behavior from the 15.03 code base. For the better yes, but some users may have relied on the updated simulation data without knowing it.

Overall, either of these options will improve the user experience when printing a rocket.

With the current behavior, it takes approximately 12.5 seconds to generate the DesignReport. With the proposed behavior this takes only 279 ms when choosing to use current simulation data and 10620 ms when using the SimulationRunDialog (its multi-threaded).

@teyrana @itshanney @neilweinstock - any preferences?

neilweinstock, 2020-04-27:
FWIW I can't reproduce this problem.  For me, Save as PDF for the Boosted dart example is nearly instant, either on 15.03 or latest unstable.

In general, I would prefer to avoid silently changing default behavior from 15.03.  So if you're going to choose option 2, I would at least add a warning that existing simulation data will be used if they're known to be out of date.

Other than that, I'm not sure I have a strong preference.

wolsen, 2020-04-27:
@neilweinstock interesting, how long does it take to run the simulations for you on the Boosted dart?

neilweinstock, 2020-05-01:
Between 2 and 3 seconds.  Here are my settings:
[image]

Running on a 6.5 year old iMac which was fast when I bought it (!) but certainly no speed demon by modern standards.

PR #665 (2020-05-24): [Fixes 637] Update print dialog to allow simulation control
Updates the print dialog to allow for simulations to be re-run
or not. Selecting the "Update simulation data" checkbox will
re-run any out of date simulations prior to generating the
design report.

Fixes #637
commit b673a591 (2020-05-24): Update print dialog to allow simulation control

Updates the print dialog to allow for simulations to be re-run
or not. Selecting the "Update simulation data" checkbox will
re-run any out of date simulations prior to generating the
design report.

Fixes #637
closed 2020-05-24 by teyrana