openrocket/openrocket #640: Edit dialog doesn't close with the "close window" titlebar widget

neilweinstock, 2020-05-01:
In the current build, the edit dialog will not close with the "close window" titlebar widget, either the red stoplight button on Mac or the X button on Windows.  It just ignores any clicks to those widgets.

It *will* close if you click the "close" button in the lower right, though.

Tested on the latest unstable, based on commit 1c7171e, both on Mac and Windows.

teyrana, 2020-05-02:
Suspect Commit
-------------
https://github.com/openrocket/openrocket/commit/336b899a066f861f021e346e27dec84dd7e7da49

neilweinstock, 2020-05-02:
What does "Suspect Commit" mean?  I haven't seen that before.

closed 2020-05-02 by teyrana

teyrana, 2020-05-04:
@neilweinstock 
Hah, in other words:   "this is the commit which introduced a change, which looks like it's related, and maybe possible caused the error that you're complaining about.   Some one should go take a look at this commit, because it'll probably clue you in to what the change is"

neilweinstock, 2020-05-04:
Got it. 😊

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

commit 8288de65 (2020-05-02): Merge pull request #646 from teyrana/640-fix-dialog-close

[fixes #640] Dialogs should now close with the [x] button
commit d038b1b1 (2020-05-02): [fixes #640] Dialogs should now closing with the [x] button in the corner.  (and 'close' button.  And ESC)