openrocket/openrocket #641: CG shown off-axis

neilweinstock, 2020-05-01:
CG is sometimes shown off-axis in 3D Finished view.  See attached design.

Side view:
[image]

3D Finished view:
[image]

Tested on latest unstable build.

[cg_render_bug.ork.zip](https://github.com/openrocket/openrocket/files/4562127/cg_render_bug.ork.zip)

teyrana, 2020-05-02:
Notes: 
1. iirc, The 2d view ignores the y,z components of the CG and CP: 
[RocketPanel.java#L634](https://github.com/openrocket/openrocket/blob/8288de651f19a0ad7fee0f82770a0c87b344df4e/swing/src/net/sf/openrocket/gui/scalefigure/RocketPanel.java#L634-L635)

neilweinstock, 2020-05-03:
Interesting.  Is it a bug that the 2d view ignores Y and Z?  I'm not sure why it should.

JoePfeiffer, 2020-05-04:
I'm guessing it should not.  CP is forced to be on-axis because its calculation with individual fins in a FinSet (instead of calculating the FinSet as a whole) erroneously calculates an off-axis CP (I've spent some time this morning trying to find where we concluded that, and haven't been able to locate the discussion nor the PR that does it.  I know I'm just looking in the wrong place).  I'll hazard a guess that before we forced the CP calculation, the side view rendering did it instead and what Dan is finding there is an artifact.

CG off axis seems like it has to be an error in the calculation, which was being masked in the side view rendering.

(all this is speculation and memory, of course)

teyrana, 2020-05-04:
> Is it a bug that the 2d view ignores Y and Z? I'm not sure why it should.

I interpret it as a bug, certainly.   It's an optimization which makes sense when everything is on one axis, but it should be updated. 

Why?:  it's simply ignored at display:  check it out:
https://github.com/openrocket/openrocket/blob/8288de651f19a0ad7fee0f82770a0c87b344df4e/swing/src/net/sf/openrocket/gui/scalefigure/RocketPanel.java#L634-L635

I can add that particular change to the PR fixing this issue.

commit 5e264c86 (2020-05-07): Merge pull request #649 from teyrana/641-fix-cg

[fixes #641] MassCalculator now accounts for rotations in the tree
closed 2020-05-07 by teyrana