openrocket/openrocket #642: Slight rendering bug on zoom control on Mac

neilweinstock, 2020-05-01:
The zoom widget has a slight rendering problem on the Mac: the pull-down handle is shown slightly above the value field.  This screenshot shows it on the main window:

[image]

The same problem is also in the scaling controls in the lower left corner of the freeform fin shape editor.

Noticed on the latest build, not sure how long it's been there.

wolsen, 2020-07-13:
Looks to be related to https://bugs.openjdk.java.net/browse/JDK-8179076.

A comment added to that bug lists: "Workaround is to move and resize to fit as required"

neilweinstock, 2020-07-13:
Sure does!

This one really is minor as can be, I wouldn’t spend too much time on it.

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

hcraigmiller, 2021-06-14:
@neilweinstock: Is this still an issue?

neilweinstock, 2021-06-14:
Yes.

SiboVG, 2025-02-09:
Fixed with the new FlatLaf UI.

closed 2025-02-09 by SiboVG