openrocket/openrocket #643: Two button issues

neilweinstock, 2020-05-01:
[image]

1) Why do the "New Stage", "New Boosters", and "New Pods" buttons have the word "new" in them?  *All* the buttons create a new component, that is their function.  Propose removing the word "new" from these three button names.

2) I'm pretty sure the "Mass Objects" section should start on a new row.

Tested on latest unstable.

JoePfeiffer, 2020-05-01:
> 1. Why do the "New Stage", "New Boosters", and "New Pods" buttons have the word "new" in them?  _All_ the buttons create a new component, that is their function.  Propose removing the word "new" from these three button names.
Excellent question -- unless someone comes up with a reason they're different, I agree

> 2. I'm pretty sure the "Mass Objects" section should start on a new row.

Agreed.  It also makes sense for it to come immediately after the "Inner component" row, before the "Assembly Components" (since "Assembly Components" has a markedly different "feel" than the other components).

Hmmm, while I'm looking at the panel, the capitalization and singular/plural should be made uniform.

neilweinstock, 2020-05-01:
Agreed, except for the singular/plural issues.  I *think* they're OK as is.  It's subjective.

JoePfeiffer, 2020-05-01:
I don't care whether they're singular or plural, just so they agree.  If "Mass objects" should be plural, "Inner component" should be as well (or both should be singular).

neilweinstock, 2020-05-01:
Oh yes, for category names I agree. I was looking at the individual button names, which have reasons to be different.

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

JoePfeiffer, 2020-05-01:
Got it.  Yes, for individual button names, I agree.

teyrana, 2020-05-02:
(a) these are two separate issues
(b) regarding the plurality issues: They're different, because they add different numbers:
     - adding a new mass object adds a single mass object.
     - adding a new pod-set adds 2 bilateral pods, _by default_ --  in the same way that adding fins adds 3 fins, by default. 
(c) if we're going to rearrange the order of the buttons, the assemblies should be up top, because they need to be added first, and other components are added _to_ them.

teyrana, 2020-05-02:
And while we're comparing the buttons, let's look at on a monitor with more pixels: 
[image]

See all that blank, unused space in the upper right?   The design intent of continuing the buttons after the previous section finishes is to pack the "new component" buttons into a smaller space. 
If the buttons take up less space (or more efficiently, that means more viewable area for the rocket itself. 

Arguably, an even better change would be for the buttons to re-flow based on the available panel real estate -- i.e. automatically adapt.   Aka responsive design.

neilweinstock, 2020-05-02:
I have a large monitor but don’t typically work with OR in a very large window. That should not be a requirement in order to make the interface usable.  A responsive design would certainly be better, but that doesn’t sound like the sort of thing we should start messing with at this point in the release.  It’d be a good thing to put on the Todo list, though

I would also advocate for smaller buttons that can pack in more tightly.  I believe they are needlessly large and spread out right now.  This too would be something for a future release.

(b) regarding the plurality issues: They're different, because they add different numbers:
- adding a new mass object adds a single mass object.
- adding a new pod-set adds 2 bilateral pods, by default -- in the same way that adding fins adds 3 fins, by default.

Yes, we’re all in agreement here; there was a misunderstanding earlier.  It’s the pluralization of the *categories* that Joe was talking about, which pretty much just boils down to “Inner Component” should be “Inner Components”.

(c) if we're going to rearrange the order of the buttons, the assemblies should be up top, because they need to be added first, and other components are added to them.

Well I would certainly say that they should be first or last, not in the middle, since they are qualitatively different from everything else.  Although they might occasionally be used first, they are more often not used at all, or at least infrequently.  But I wouldn’t be averse to putting them up top, if they’re on their own row.

Optimal button layout is not an easy problem.

  *   Neil

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

JoePfeiffer, 2020-05-02:
Neil is right, this isn't the time to start thinking about moving to a responsive design.  The buttons could definitely be shorter; the only ones with icons dictating the current height are "New stage" and "New Boosters", so those would need to be modified.

Note the category names should be consistent in both pluralization and capitalization -- so we also want "Body Components and Fin Sets" (or possibly just  "External Components"?), "Inner Components", and "Mass Objects".  Note that while the pluralization of the components seems fine, their capitalization isn't consistent.

Assembly Components should come either first or last.  Dan is right that workflow indicates they should be first; Neil is right that frequency of use indicates they sould be last.  I've got a weak preference for last, but would not be terribly upset with first.

neilweinstock, 2020-05-02:
An advantage to having them first is that folks using the new version would find them quickly, rather than maybe not noticing them at the end of the list.

I don’t know if that’s a good reason to put them there, but I do think it’s a real issue.

  *   Neil

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2020-05-08:
> Optimal button layout is not an easy problem
QFT.  :/

> An advantage to having them first is that folks using the new version would find them quickly, rather than maybe not noticing them at the end of the list.

I totally agree with this design principle.   I'm glad we're on the same page there :)

 > . . . this isn't the time to start thinking about moving to a responsive design.

Well, the issue is already submitted, so that particular cat is already out of the bag.    We're already talking about it, and the talking takes the most time for these kind of things.

*Besides* which, it already _is_ responsive.  We're just talking about _in which way_ should it be responsive.   (don't believe me?   just change the window size -- the buttons reflow to adjust.  That's all I meant.)

> I have a large monitor but don’t typically work with OR in a very large window. That should not be a requirement in order to make the interface usable. 

I think we may be crossing wires here.   A large monitor is merely supported; as are smaller monitors / window sizes.   And having taken a look how the buttons reflow for smaller windows, I see what you mean and I agree. 

> Assembly Components should come either first or last.  .... workflow indicates they should be first; . . . frequency of use indicates they should be last.

I would also add that "first" is also optimized for new users, and would tend to head-off questions like "where did the 'new stage' button go?"

teyrana, 2020-05-08:
Partial PR: #659, which addresses the alignment / layout issues.

JoePfeiffer, 2020-05-08:
> _Besides_ which, it already _is_ responsive. We're just talking about _in which way_ should it be responsive. (don't believe me? just change the window size -- the buttons reflow to adjust. That's all I meant.)

This is what I get for not verifying my memory before responding -- I could have sworn it went to a horizontal scrollbar when the window was resized.  Obviously I remembered wrong.

> I would also add that "first" is also optimized for new users, and would tend to head-off questions like "where did the 'new stage' button go?"

Excellent point.

neilweinstock, 2020-05-08:
I think the likely "next level" of responsiveness would be put a button group on a new row only when it could not fit completely on the current row.  In that case I would also add an outline or something around each group.  Sort of like the way toolbar button groups work in apps with older-style toolbars (vs. MS ribbon-style).

Would be a reasonable thing to fiddle with for a future release, along with smallifying the buttons a bit. :)

I think Daniel's latest PR looks good for this release, at least with regard to button group placement.

teyrana, 2020-05-09:
@JoePfeiffer @neilweinstock 

If I understand the consensus correctly, I've made the text label changes in #661 
(w/screenshot)  Take a look if you like.

neilweinstock, 2020-05-09:
Looks good to me!

PR #659 (2020-05-09): [ui] Moves new assembly buttons to their own row in the New Component panel.
This PR addresses, but does not completely fix #643.

[image]
closed 2020-05-09 by teyrana