openrocket/openrocket #648: Error in calculating CP with phantom body tubes:

teyrana, 2020-05-02:
CG is sometimes shown off-axis in 3D Finished view.  See attached design.

Related to #641, but this issue is for CP calculations.

Side view:
[image]

3D Finished view:
[image]

Tested on latest unstable build.

[cg_render_bug.ork.zip](https://github.com/openrocket/openrocket/files/4562127/cg_render_bug.ork.zip)

If the pod-fins are offset *past* the end of their parent bodies (Offset via BOTTOM: x > 0) ... CP is NaN.

If the pod-fins are offset *past* the end of their parent bodies (Offset via BOTTOM: x > 0) ... CP  is reasonably.

neilweinstock, 2020-05-04:
Here's a design I cooked up yesterday that shows the CP off-axis.
[image]
[retro_lander.ork.zip](https://github.com/openrocket/openrocket/files/4574958/retro_lander.ork.zip)

JoePfeiffer, 2020-05-04:
These CP bugs are like roaches, aren't they...

> If the pod-fins are offset _past_ the end of their parent bodies (Offset via BOTTOM: x > 0) ... CP is NaN.
> 
> If the pod-fins are offset _past_ the end of their parent bodies (Offset via BOTTOM: x > 0) ... CP is reasonably.

These look like you're describing the same case?

teyrana, 2020-05-04:
*sigh* yea.  There *is* a lot of them.

PR #652 (2020-05-06): Avoid considering cp with 0 weight when calculating worst-case CP for display in RocketInfo
This fixes #648 (errrors in CP calculation).  The fins in the first example design in the issue are extremely small, and at some angles of attack can result in a CP weight so small as to be meaningless.  This PR only considers CP's with a non-zero weight in finding the worst case for display in the RocketInfo panel.

Also, in RocketPanel, a comparison was being made between CP weight at worst case and rocket minimum mass.  These have nothing obvious to do with each other, so this PR changes the test to  [... cut]
closed 2020-05-06 by teyrana