openrocket/openrocket #653: Display CP weight?  Flag when too small?

JoePfeiffer, 2020-05-05:
At present, we display a rocket's CP and stability margin but don't display the CP's weight.  This means a rocket could appear to be stable when the fins are actually so small they won't apply any useful restoring force.  Should we find some way to indicate to the user what the weight of the CP is?  Should we find some heuristic that uses both CP location and weight (ie the actual moment) to flag a rocket likely to be unstable?

Certainly not for this release, but worth considering in the future.

teyrana, 2020-05-08:
Correct me if I'm wrong, but is this a value people actually use in rocket design? 

I'm trying to come up with a counter-example, where a rocket is "barely" stable (by, say, less than 1 caliber) ... but the magnitude of the restoring force matters?    And where that magnitude is for some reason _not_ already represented in the Cp.x?

JoePfeiffer, 2020-05-08:
My guess is it's baked in to heuristics we use, both formally and informally -- we want our fins to look "big enough" (regardless of what the CG-CP looks like), and sometimes people will mention fin height should be at least one caliber.

I'm also wondering if the "minimum one caliber stability" rule is really a simplification of a "need adequate restoring force" rule.

A thought to consider *after* this release is out!

neilweinstock, 2020-05-13:
This could be useful.  There are instances where rockets are technically stable but have inadequate restoring force. For instance, imagine a simple rocket with a lot of nose weight but very small fins.  These rockets will tend not to fly very well.

I would agree this would be interesting to look into in the future.  What I don't know is how to present the number to the user in such a way that they know what to do with it.

neilweinstock, 2020-05-13:
> I'm also wondering if the "minimum one caliber stability" rule is really a simplification
>of a "need adequate restoring force" rule.

I don't think it's that simple, but I'm having a hard time putting it into words because I haven't studied control systems in... oh... too long. :)

teyrana, 2020-05-17:
> I don't think it's that simple, but I'm having a hard time putting it into words because I haven't studied control systems in... oh... too long. :)

Well there's no need to bring control systems into it.   It's a simply equation of forces and moment arms.   Fins assume some angle of attack (AoA)(== alpha)  where the rocket is not on-center to the airstream, but at that angle alpha.  Some components impart a +positive alpha (de-stabilizing) force, and some impart a negative alpha (stabilizing) force.   

The moment of a force about any particular point is simply the force times it's moment arm -- so the force and the distance have an equivalence -- large fins can be converted to a longer distance.  

And according to [NASA](https://www.grc.nasa.gov/WWW/K-12/airplane/cp.html)  The center of pressure is _by definition_ the point where those moments balance out to zero.     By definition, the center of pressure has zero magnitude.  

If you can express the "magnitude" of the Center-of-pressure,  well, that breaks basic assumptions of the Barrowman Equations.  And that would be "interesting", to say the least.

Joedang, 2020-06-05:
Here's my 2 cents... The CoM and CoP do leave out some important pieces of information: the rocket's moment of inertia and the restoring moment per angle of attack. They tell you whether the rocket will correct any angle of attack back to zero, but they don't tell you *how quickly that will happen*. 

The thing that *I* care about when designing fins/ballast is (1) whether the rocket's stable off the launch rail and (2) how it's going to wobble/cone/weathercock when launching in wind. Typically, I will just plot the angle of attack and stability margin to get that info. IMO, a nice "at a glance" equivalent to that would be the coning frequency of the rocket when it comes off the rail, which kind of answers the question "How wiggly will it be?". You could get that from the moment of inertia and the derivative of the restoring moment with respect to the angle of attack. (I'm not sure if that derivative is readily available in OR.)