openrocket/openrocket #654: CP is still shown off-center in 3D views if pods.

JoePfeiffer, 2020-05-06:
In Neil's 3fnc test rockets, CP is still shown off-center in 3fnc-take2-pods.ork.  On-center in 3fnc-take2.ork, however.

JoePfeiffer, 2020-05-06:
Just clarified the title -- the display isn't the problem (in fact, was masking the problem).

JoePfeiffer, 2020-05-06:
Ah.  The CP code for FinSets recognizes when there are three or more fins in the set, and (because they're evenly distributed around the axis) pins CP Y and Z to 0.  This doesn't work when there are pods, since there may be only a single fin on the pod, even though the pods are evenly distributed.

Need to look more closely at the moment calculations...

teyrana, 2020-05-08:
We've merged #654, so I suppose this is due for re-validation.

teyrana, 2020-05-09:
@JoePfeiffer I'm assumed you're referencing #568?

teyrana, 2020-05-09:
I just ran the 3fnc files referenced above through our `unstable` branch, and the CP is on-axis.  
It looks like this was resolved by PR #657.

Closing as resolved.

closed 2020-05-09 by teyrana

JoePfeiffer, 2020-05-09:
I wasn't specifically referencing any other issue in particular, though the examples in #568 are a good test case.  Yes, issue is resolved.