openrocket/openrocket #656: railbuttons don't rotate in 2D view

JoePfeiffer, 2020-05-06:
With the attached design, when I display in 2D and rotate the rocket the railbutton remains vertical is at moves around the body tube.  In 3D views, it rotates appropriately.
[railbutton-doesn't-rotate.ork.zip](https://github.com/openrocket/openrocket/files/4589104/railbutton-doesn.t-rotate.ork.zip)

teyrana, 2020-05-08:
## At 0°
[image]

## At 90°
[image]

JoePfeiffer, 2020-05-08:
At 90 it should look like a disk (just as fins are shown in an orthographic view).  Also, at a 180 rotation the button is shown inside the body tube.
[image]

teyrana, 2020-05-08:
see #660

commit 247bbbc6 (2020-05-09): Merge pull request #660 from teyrana/656-fix-button-render

[fixes #656] Rail Buttons now properly display in 2d side views
closed 2020-05-09 by teyrana